ka9q/ka9q-radio #1: Include missed on attr.c

srsampson, 2022-04-14:
In attr.c it has:

```
#ifdef __linux__
#include <attr/xattr.h>
#else
#include <sys/xattr.h>
#endif
```
But on Ubuntu the include is in the ```sys``` directory. Changed to:

```
#include <sys/xattr.h>
```

Maybe other linux versions differ, which will complicate the #ifdef

ka9q, 2022-04-15:
That's weird. The man page now says

#include <sys/types.h>

#include <sys/xattr.h>

Maybe it changed after I originally wrote attr.c. I hate this sort of 
thing. Anyway, I've made the change. Thanks for the report! --Phil

closed 2022-04-15 by srsampson

ka9q, 2022-04-16:
well, that wasn't very hard...

srsampson, 2022-04-16:
> well, that wasn't very hard...

I'm just glad I wasn't some special case! Thanks.