ka9q/ka9q-radio #7: packet misses most APRS packets

argilo, 2022-04-18:
I'm running the packet program with the `-v` option to watch APRS packets, and it looks like only about 25% of packets transmitted by my local digipeater (which is less than 1 km away) are printed.

Before I dig into this, I wondered whether you might already know the cause @ka9q.

ka9q, 2022-04-18:
Not offhand. I run this as part of a receive-only i-gate and haven't 
noticed any problems that can't be explained by weak signals. But I've 
never quantified its performance, so it's entirely possible that it's 
subpar.

argilo, 2022-04-18:
OK. I'll have a look at it soon and let you know what I find.

argilo, 2022-04-18:
I had a look at some recordings made with pcmrecord, and it seems like radio's squelch is cutting off the ends of the packets.

argilo, 2022-04-19:
I increased `squelchtail` in fm.c from 0 to 1, and that seems to have fixed it. Packet tails are no longer chopped off, and all 10 of the most recent packets decoded correctly.

ka9q, 2022-04-19:
I've been thinking of making that a config file option.

ka9q, 2022-04-19:
I introduced that squelchtail mechanism to fix precisely this problem 
some time ago, and I guess I forgot to do enough regression testing when 
I later tweaked it to tighten up audible squelch tails on non-data signals.

It has to do with how the squelch is either open or closed during each 
20 ms (typical) block.

I think it's (almost) another classic off-by-one error. The SNR is 
computed as an average over a block. So if a signal just above the 
squelch closing threshold disappears during the block so the average SNR 
for the block falls below the closing threshold, the whole block will be 
squelched, chopping off (on average) 10 ms of actual signal.

I should make squelchtail a configuration option so it can be kept tight 
on voice channels without clipping off the tails of packet 
transmissions. I already configure voice and data differently by 
enabling de-emphasis only on voice (this is a recent change).

ka9q, 2022-04-19:
OK, I've created a "squelchtail" config parameter in modes.conf with a 
default of 1 frame.

This default is used in modes.conf by the FM and FMN entries, i.e., flat 
FM for digital modes like packet.

The PM and PMN entries set squelchtail=0 to eliminate audible squelch 
tails on analog voice channels.

NBFM with pre- & de-emphasis is really phase modulation above the 300 Hz 
corner frequency, hence the PM name.

I also cleaned up the modes.conf file, listing all the options and their 
defaults in comments at the end.

Give it all a try. --Phil

argilo, 2022-04-19:
Looks good! Packets are no longer getting cut off.

closed 2022-04-19 by argilo

ka9q, 2022-10-11:
Interesting. I had that problem a while ago, and put logic into fm.c to 
hold the squelch open for a configurable number of frames after the SNR 
drops below a threshold. Maybe that needs tweaking.

argilo, 2022-10-11:
The change you made back in April (94667e614610a8ddacb0b787f18e46ad842fe122) did fix the problem for me.