ka9q/ka9q-radio #10: buffer overflow in control

argilo, 2022-04-22:
When I run `control`, it immediately exits with the following error:
```
*** buffer overflow detected ***: terminated
Aborted (core dumped)
```
It appears this happens because `Frontend.input.status_fd` is -1 when it is used in `FD_ISSET` on line 519:

https://github.com/ka9q/ka9q-radio/blob/b667f885bd93be7cbccafab65686ab41cef1d3c2/control.c#L519

It seems to be timing-dependent, since the problem vanished when I attempted to investigate the problem by changing compiler settings or using valgrind.

argilo, 2022-04-22:
Adding a check prevents the crash:
```diff
diff --git a/control.c b/control.c
index ea5e191..749db64 100644
--- a/control.c
+++ b/control.c
@@ -516,7 +516,7 @@ int main(int argc,char *argv[]){
        }
       }
     }
-    if(FD_ISSET(Frontend.input.status_fd,&fdset)){
+    if(Frontend.input.status_fd >= 0 && FD_ISSET(Frontend.input.status_fd,&fdset)){
       // Message from the front end
       unsigned char buffer[8192];
       struct sockaddr_storage sender;
```
I don't understand the program well enough to know whether that's the correct fix though.

ka9q, 2022-04-27:
Good catch. Not sure why it happens because up on line 435 I set the bit in fdset only if Frontend.input.status_fd is not -1. Your fix can't cause any harm, but I want to study the code to figure out how it happens.