ka9q/ka9q-radio #19: Added support for SDRplay (plus some multicast and minor fixes)

fventuri, 2022-07-07:
This is not really an issue; please feel free to close it anytime.

I thought it might be a useful addition to also have a front-end for the SDRplay RSP devices ('sdrplayd'); while working on it I also fixed a few things with multicast and the Avahi client code, so it should now consistently support a default interface for multicast (or an interface specified as the suffix ',iface' at the end of the DNS name).

I also found and fixed a possible problem where it looks like the software LO was not being initialized (i.e. 'set_osc()' was not being called) when the initial value of 'remainder' is 0, so I added a call to 'set_osc()' in main.c.

I ran a quick test here and I was able to listen to a local FM broadcast station using the programs 'control' and 'monitor'.

You can find the code in this repo: https://github.com/fventuri/ka9q-radio - to make it easier to look at those individual changes and enhancements, I created several branches, where each one of them contains the commits for that specific change; for instance the branch 'sdrplay' contains just the code I changed to add the front-end for the SDRplay devices, and so on. I also created a branch called 'all-changes' that has exactly the code as I am running it here.

Let me know what you think; if you would like to integrate some/all of this changes into your main repo, I'll be happy to help.

Regards,
Franco Venturi K4VZ

ka9q, 2022-07-07:
This looks great! I have a few SDRplay devices but I haven't used them 
much because the hardware API is proprietary.

Re set_osc() I remember looking at that just a few weeks ago. I was 
trying to streamline the code by removing tests in low-level code, and I 
seem to recall concluding that an oscillator would implicitly initialize 
itself to 0 hz if it wasn't explicitly set.

I'm not very familiar with git so I'll have to figure out how to merge 
your code into mine.

Thanks again!

Phil

ka9q, 2022-07-07:
Okay, I looked at the fine oscillator initialization. I remembered 
correctly; I looked at the osc code pretty closely a few weeks ago to 
ensure that an oscillator was implicitly initialized to zero hertz and 
zero phase as long as the entire osc structure was zeroed. This is the 
case at startup since it's part of struct demod, which is zeroed in 
radio.c/alloc_demod().

If step_osc() is called on a zeroed osc structure, the call to 
renorm_osc() will happen immediately. It will see that the current phase 
vector is not initialized and set osc->phasor, osc->phasor_step and 
osc->phasor_step_step to unity (1 + j0). This implicitly initializes the 
oscillator to zero phase and zero frequency. But your point is very well 
taken: it should be clear to anyone reading the code that the oscillator 
is properly initialized.

When you execute your addition to main.c:

set_osc(&demod->fine,demod->filter.remainder/demod->output.samprate,demod->tune.doppler_rate/(demod->output.samprate 
* demod->output.samprate));

demod->filter.remainder and demod->tune.doppler_rate are still zero, so 
you might as well call it as

set_osc(&demod->fine,0,0);

but your line will work, so I've added it. Thanks!

--Phil

Message ID: ***@***.***>

ka9q, 2022-07-07:
Thanks much for your look at my handling of multicast interfaces. That 
stuff was quite unpolished, and I admit I hadn't spent a lot of effort 
on it because most of my systems just use the default. But I've been 
meaning to look at it to solve some problems on systems with non-default 
network interfaces, such as my Macbook when I use an plug-in wired 
Ethernet adapter.

I updated main.c in my own code with your version. This is probably the 
wrong way to do it; in my experience the simplest, most intuitive and 
obvious way to do *anything* in git is always the wrong way (or so I'm 
told...)

Phil

Message ID: ***@***.***>

ka9q, 2022-07-07:
I've added your changes to radio.c to handle 16-bit little endian IQ 
streams.

I had previously used the IQ_PT type for a now-obsolete format that had 
a metadata header at the front of every packet. I haven't used that 
format for a long time so I don't mind reusing that type for a 
headerless format (which I assume you need for the sdrplay).

Thanks!

Phil

Message ID: ***@***.***>

ka9q, 2022-07-07:
Regarding your changes to avahi.c:

It looks like you copy the service_name (and then dns_name) up to the 
comma, if it exists, and then copy the null from the end of the argument 
string to the destination.

Wouldn't this be more straightforward?

if(service_name){

     userdata->service_name = strdup(service_name);

     char *ifp = strrchr(userdata->service_name,',');

     if(ifp)

         *ifp = '\0';

}

strdup() allocates space with malloc, so it can be properly freed even 
if a null has been written into the middle of it.

Phil

WhyMessage ID: ***@***.***>

ka9q, 2022-07-07:
Oh, I think I see what you're doing. The service name will be something like

"2m radio (2m.local)"

and if an interface is present it will be

"2m radio (2m.local,eth0)"

and you're trying to retain the trailing paren after removing the string 
",eth0". Is that right?

I think this highlights some pretty kludgy features of my interface 
naming scheme, and I need to go figure out how to do it all far more 
cleanly. This is why I hadn't really tried to make things work properly 
with non-default multicast interfaces. I think I wanted to see whether 
that was all really needed before I invested a lot of time into it.

I didn't start using libiniparser until a few months ago, and until then 
I was overloading a lot of options onto things like DNS names that I 
shouldn't have been.

I'm thinking it might now be a lot cleaner to just do away with the 
,iface notation entirely. Explicitly specify a non-default interface in 
the config file. You've already added the hook to do this. What do you 
think?

Thanks.

Phil

ka9q, 2022-07-07:
Now I remember why I put my SDRplay boxes on the shelf and never really 
tried to use them.

The required library sdrplay_api is not a standard package that you can 
just install in a few seconds with 'apt get' on Debian Linux on the x86 
or on the Raspberry Pi. You have to download it through a bunch of funky 
forms on their website that require the device serial number, your 
current shoe size and your first-born's social security number. And if I 
recall correctly, it's object code only. I don't consider that acceptable.

If someone wrote an open source driver for this thing I'd be much more 
willing to use it. The SDRPlay hasn't been a high priority for me since 
I discovered the Airspy devices, which do provide open source library 
modules in the standard Linux distributions. Maybe they'll relent if 
enough developers take a stand.

I've added your sdrplayd.c (and .conf and .service) files to my 
repository (thanks!) but for now I've commented them out in 
Makefile.linux so it's not built or installed by default.

I'm thinking it's time to break up ka9q-radio into separate projects so 
people don't have to build everything just to get a few components. 
After all, I tried to design it to be modular. Comments?

Phil

Message ID: ***@***.***>

closed 2022-07-09 by fventuri