ka9q/ka9q-radio #22: compute_tuning() question in radio.c

srsampson, 2022-08-06:
Concerning the AirspyR2 I see mention of a 10 Mhz bandwidth, but then it says it is limited to 9 MHz to have no aliasing.

The 'r' value passes if less than 125,000 which would be what you would get for a (frontend - tune value) of 10 MHz.
(well, if my math is correct): i.e., N = 250000, M = 50,001, samplerate = 10000000

Code in question:
https://github.com/ka9q/ka9q-radio/blob/main/radio.c#L646

I was wondering if this should be:
```
if (abs(r) >= N/2)
```

To prevent aliasing.

ka9q, 2022-08-06:
Are you asking only if it should be a greater-than-or equals instead of 
a greater than? I don't think one FFT bin (typically 40 Hz or so) makes 
much difference, and there's some ambiguity about signals at exactly the 
Nyquist rate (Fs/2).

This doesn't really affect aliasing, because the code in filter.c 
automatically zeroes out a downconverter's input frequency bins that 
extend beyond the current edges of the front end's coverage to prevent 
aliasing.

The only purpose of this test is to tell the digital downconverter 
whether it should just go completely idle and wait for the front end to 
be retuned back into range rather than apply the bandpass filtering and 
execute the rest of the demodulator on frequency bins that are all 
zeroes. I.e., it's just a CPU performance enhancement.

E.g., I have a statically-tuned virtual channel for WSPR on every HF 
band, but my AirspyHF+ can only cover ~912 kHz at a time so that when I 
manually tune a separate receiver channel around 40m, say, the fixed 40m 
WSPR channel will activate and decode anything it hears. If I manually 
tune to 20m, then the 40m WSPR channel will go dormant and the 20m WSPR 
channel will wake up.

Some day I'll have a front end that can cover all of HF simultaneously, 
then I won't need this dormancy feature anymore and I will be able to 
demodulate WSPR on every band simultaneously.

srsampson, 2022-08-06:
Thanks for replying, and appreciate the code description. I hesitated to even mention this, as I know you're dealing with multiple SDR types, and after-all, the code works fine. Sometimes, in reading code, I generate questions for myself, so this was just one where I wasn't sure. I'll go ahead and close this.

closed 2022-08-06 by srsampson