ka9q/ka9q-radio #27: Support for hackrf

pinkavaj, 2022-10-16:
It looks like `hackrf` support is not implemented.

* There is a `hackrf@.service`
* There is relevant line in `Makefile.linux`

but it looks like `hackrf.c` does not exists. The implementation is missign right?

Btw would make more sense to implement it specifically for HackRF or just use somethink like `soapysdr`?

ka9q, 2022-11-11:
I have a HackRF, and I used to support it. But I haven't used it in a while, and I haven't kept the front end module up to date. If you'd like to reimplement it using soapySDR, that would be wonderful -- I could use it as a template for other front ends.

ncorgan, 2023-05-01:
https://github.com/ncorgan/ka9q-radio/tree/soapy-support

I started a general SoapySDR branch a while back, but I haven't touched it in a while because of life stuff. If there's interest in it, I can get back on it. I don't have a HackRF myself, but I have an RTL-SDR, a USRP, and a few Lime devices, so I can test a few frontends.

Phil, it sounds like you're looking for hardware-specific support through SoapySDR rather than general support?

ka9q, 2023-05-01:
Yes, if you can provide SoapySDR support I would love to integrate it. It probably doesn't provide all the features I want (like exact tuner frequency readback) but it would be very handy in supporting hardware I don't already support natively.

tavi343, 2023-05-14:
I would like to use RX888-MK2 with soapy_power to get a wide band spectrograph for solar radio bursts recording. It is cheaper than USRP or Lime SDRs. I would replace the analog TV tuner CALLISTO instrument (see e-callisto.org). The most interesting band is between 20-80MHz on HF but a wider band with fast scan is better. No need to decode anything, just record the power vs frequency. I've try with Airspy R2 and bladeRF X40+XB200 and these are too low speed. We need more than 10 full spectrum/second.

ka9q, 2023-09-10:
ka9q-radio performs large FFTs on its A/D input stream, not primarily for spectral analysis but as part of a filter bank technique called _fast _correlation__. I'm already adding spectral analysis to drive waterfall displays and to enable search-and-decode applications. So the underlying code you probably need is already there. I'm working on the command-line programs to extract it.

The RX-888 MkII is a wideband SDR front end that works in two modes: direct sampling at up to 130 Ms/s, and a downconversion mode that works up to the middle of L-band (1-2 GHz). The downconverter uses the very common R820/828 digital TV tuner so it's limited to about 8 MHz at a time. So far I've only implemented the direct sampling mode since it's easily the most cost-effective device we've found so far for capturing the entire HF spectrum at once. 

It can actually go up as far as the 6 meter amateur band (50-54 MHz), so it'll cover part but not all of the 20-80 MHz range you're interested in. But the stock unit has thermal problems so most of us are running it at half rate (~65 MHz), especially since we're mainly interested in HF. This requires an external 30 MHz low pass filter to suppress aliasing.

ka9q, 2024-04-03:
Check out G0ORX's web-based spectral display for ka9q-radio: https://github.com/fventuri/ka9q-web
It's still pretty rough but it does use my "spectrum" pseudo-demodulator to drive the spectrum analyzer and waterfall.

ka9q, 2025-01-14:
I've pushed up a working SDRPlay subsystem. Go see the recent commit log and Makefile.linux for the details of what I did. I've had an RSP1 and RSP2 on my shelf for years unused until now. Anybody who wants to contribute similar drivers for the Fobos, Hackrf, etc, is more than welcome.

closed 2025-01-14 by ka9q