ka9q/ka9q-radio #39: Compile error on Ubuntu

berrySkies, 2023-09-09:
Hi, Just tried to install ka9q-radio for first time on a Ubuntu system. See attached screenshot for make error messages. 
[image]

Thanks for any assistance. 
Regards Alan VK2ATW

ka9q, 2023-09-10:
Ah, I caught that one myself and fixed it last night. Getting all the #includes right can be a problem. On some systems, one include file may automatically include the one you need, masking the fact that you didn't include it explicitly. And it may compile on some systems and not others.

closed 2023-09-10 by ka9q

berrySkies, 2023-09-10:
Thanks Phil and THANKYOU for making the software available. I'm looking forward to learning it's use and experimenting. Best wishes from down under. Alan