ka9q/ka9q-radio #40: FT4 decoder?

azwirko, 2023-10-02:
Are there any plans for an FT4 decoder?

Thanks & 73

andyz - K1RA

ka9q, 2023-10-10:
Eventually. Would you like to work on it? ka9q-radio is very modular, so you can add one without having to get into my code at all. You don't even have to link to it, you just read an output stream from a receiver channel on the correct frequency.

azwirko, 2023-10-10:
Hi Phil
 I'd like to get this working, but was confused why you'd written your own FT8 and WSPR decoders and not developed a mechanism to leverage third party standard decoders like WSJT-X's jt9 executable which supports command line decoding of JT4/9/65, FT4/8, FST4/W, Q65 via WAV file or shmem.  Quite awhile ago I'd worked with Pavel Demin on his Redpitaya code on ARM and created a means to hand WAV files off to the jt9 decoder running on ARM.  If this could be done through your architecture I'd like to pursue that instead of rewriting an actual FT4 decoder like you did for FT8.

Thoughts, suggestions on where to start?

Thanks & 73

andyz - K1RA

azwirko, 2023-12-07:
Phil

I think you can close this issue.  I've got a solution I'm working on and running discussed here:

https://groups.io/g/NextGenSDRs/topic/new_qra_skimmer_for/103044298

73

andyz - K1RA

closed 2024-03-01 by ka9q