ka9q/ka9q-radio #44: Build error - Makefile.linux missing wspr-decoded

lukyduck, 2023-10-20:
Trying to compile and install on Ubuntu 22.04  I get a Make error that wspr-decoded is not found (but it is specified in the EXECS line). It looks like this may be in state of flux, as I see comments that wspr-decoded was merged with jt-decoded. 

Is there a needed update in the Makefile.linux?
Can I just remove the wspr-decoded from EXECS?

Excuse me if I am just not understanding the Makefile. No expert in that area

73,

Tom, KG3V

lukyduck, 2023-10-25:
This has clearly been addressed in the latest Makefiles, so this can be CLOSED. No need to waste your time.

The radio package works nicely with Airspy-hf on my laptop running UBUNTU 22.04

73,
Tom, KG3V

ka9q, 2024-03-01:
Glad it works!

closed 2024-03-01 by ka9q