ka9q/ka9q-radio #45: Prevent Multicast over LAN

azwirko, 2023-12-07:
I'm attempting to run ka9q-radio completely local on an Ubuntu PC and desire NOT to multicast out over my wired LAN.  I read there is a "iface" setting in the global section of radiod.conf which I've used, as follows:

[global]
iface = lo

But sniffing my LAN I still see traffic go out over the main ethernet interface from my private 192.168 host IP.  

I've found someone suggest turning off multicast on my main interface using "sudo ifconfig eno1 -multicast" and ensure its enabled on localhost "sudo ifconfig lo multicast" resulting in:

eno1: flags=67<UP,BROADCAST,RUNNING>  mtu 1500
        inet 192.168.2.228  netmask 255.255.255.0  broadcast 192.168.2.255
        ether b0:4f:13:00:c0:a5  txqueuelen 1000  (Ethernet)
        RX packets 32472  bytes 22294178 (22.2 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 749027  bytes 396046200 (396.0 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  memory 0x92f00000-92f20000  

lo: flags=4169<UP,LOOPBACK,RUNNING,MULTICAST>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 1798  bytes 316424 (316.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1798  bytes 316424 (316.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

But restarting radiod service still pushes multicast data out eno1 onto my LAN.  Has anyone successfully used "lo" and ensured they aren't really still sending over their LAN?

Suggestions where to look or what to try next?

Thanks & 73

andyz - K1RA

rrobinett, 2023-12-07:
set ttl = 0

azwirko, 2023-12-10:
Thanks Rob, completely ignoring, commenting out using the "iface" option and setting "ttl = 0" prevented any multi-cast data from leaving my eth0 interface and reaching my LAN.

andyz - K1RA

ka9q, 2024-03-01:
Yes, that's what the ttl=0 option is for. I think I even set it to zero by default because of the problems multicast causes to "dumb" switches with slow port and especially to wifi access points.

closed 2024-03-01 by ka9q