ka9q/ka9q-radio #46: Request pcmrecord option

azwirko, 2023-12-09:
Phil
  In support of my QRA-Skimmer for your software it would help if I could pass/define a filename to your "pcmrecord" on the command line so I don't have to parse standard output and wait for you to autogenerate one.

Thanks!

andyz - K1RA

rrobinett, 2023-12-09:
Andy,

You might look at wd-record.c which listens to a PCM channel and writes out
a series of time-named wav files

Rob

azwirko, 2023-12-10:
Thanks Rob, I'll see if I can merge in pcmrecord's "-L" maxtime option to accomplish what I need to record a specific interval.

andyz - K1RA

ka9q, 2024-03-02:
'pcmrecord' handles multiple simultaneous streams on the same multicast group, so it needs to auto-generate file names to distinguish them. If you pass a single file name, how would I handle multiple streams?

ka9q, 2024-04-03:
Andy, did you still want to do this? How should I generate file names for multiple streams to the same output multicast group? Note that radiod now automatically multicasts status metadata onto each output stream so that applications like pcmrecord can use it.

azwirko, 2024-04-03:
Hi Phil
  I hacked something into the current pcmrecord for my needs back several
months ago.  I haven't had much time to revisit since. Thanks for
commenting and pointing on my QRA-Skimmer update post and some
other options for me to pursue.  I've not formalized getting QRA-Skimmer
into others hands, but my instance (K1RA-4) has been running and reporting
for 3+ months now to WSPRnet and PSKreporter.  I'll see if I can clean
things up based on your earlier input.  You can close this request.  Thanks
for your support!

73

andyz - K1RA

ka9q, 2024-12-17:
Is this issue still open? I've done a lot of work on pcmrecord recently.

azwirko, 2024-12-17:
Phil I'll take a look, but you can close this if you want.  I had to put my
QRA-Skimmer to the side over the past year (still not retired yet) and
never formalized it enough to share with the community.  Maybe I'll revisit
over the holidays and dig into looking at your code as well.

73

andyz - K1RA

ka9q, 2024-12-19:
OK. AI6VN and I wrote skimmers for FT8, FT4 and WSPR that have been operational for some time (see my reports on pskreporter.info). They log to /var/log/[mode].log, and Phil Gladstone wrote scripts that periodically upload them to his website. And of course Rob does far more analysis of the WSPR spots than just reporting. It all seems to work well, so I'm closing this.

closed 2024-12-19 by ka9q