ka9q/ka9q-radio #56: Airspy HF+ issue

HB9VQQ, 2024-06-07:
Hi, I am seeing the follwing messages

wsprdaemon@rx888:~/wsprdaemon$ sudo journalctl -u radiod@airspy.service -f -n 30
Jun 07 21:18:12 rx888 radiod[2859436]: dropped 4,096
Jun 07 21:18:12 rx888 radiod[2859436]: dropped 6,144
Jun 07 21:18:12 rx888 radiod[2859436]: dropped 10,240
Jun 07 21:18:12 rx888 radiod[2859436]: dropped 4,096
Jun 07 21:18:12 rx888 radiod[2859436]: dropped 2,048
Jun 07 21:18:14 rx888 radiod[2859436]: dropped 18,432
Jun 07 21:18:14 rx888 radiod[2859436]: dropped 6,144
Jun 07 21:18:14 rx888 radiod[2859436]: dropped 10,240
Jun 07 21:18:15 rx888 radiod[2859436]: dropped 2,048
Jun 07 21:18:15 rx888 radiod[2859436]: dropped 4,096
Jun 07 21:18:16 rx888 radiod[2859436]: dropped 34,816
Jun 07 21:18:16 rx888 radiod[2859436]: dropped 4,096
Jun 07 21:18:17 rx888 radiod[2859436]: dropped 36,864
Jun 07 21:18:19 rx888 radiod[2859436]: dropped 4,096
Jun 07 21:18:20 rx888 radiod[2859436]: dropped 2,048
Jun 07 21:18:22 rx888 radiod[2859436]: dropped 12,288
Jun 07 21:18:22 rx888 radiod[2859436]: dropped 36,864
Jun 07 21:18:22 rx888 radiod[2859436]: dropped 10,240
Jun 07 21:18:22 rx888 radiod[2859436]: dropped 4,096
Jun 07 21:18:24 rx888 radiod[2859436]: dropped 8,192
Jun 07 21:18:24 rx888 radiod[2859436]: dropped 30,720
Jun 07 21:18:24 rx888 radiod[2859436]: dropped 2,048
Jun 07 21:18:25 rx888 radiod[2859436]: dropped 38,912
Jun 07 21:18:25 rx888 radiod[2859436]: dropped 14,336
Jun 07 21:18:25 rx888 radiod[2859436]: dropped 12,288
Jun 07 21:18:26 rx888 radiod[2859436]: dropped 18,432
Jun 07 21:18:27 rx888 radiod[2859436]: dropped 18,432
Jun 07 21:18:29 rx888 radiod[2859436]: dropped 45,056
Jun 07 21:18:29 rx888 radiod[2859436]: dropped 8,192
Jun 07 21:18:31 rx888 radiod[2859436]: dropped 2,048
Jun 07 21:18:35 rx888 radiod[2859436]: dropped 18,432
Jun 07 21:18:36 rx888 radiod[2859436]: dropped 8,192
Jun 07 21:18:37 rx888 radiod[2859436]: dropped 8,192
Jun 07 21:18:37 rx888 radiod[2859436]: dropped 2,048
Jun 07 21:18:37 rx888 radiod[2859436]: dropped 24,576
Jun 07 21:18:37 rx888 radiod[2859436]: dropped 14,336
Jun 07 21:18:37 rx888 radiod[2859436]: dropped 2,048
Jun 07 21:18:38 rx888 radiod[2859436]: dropped 8,192
...

I believe this is not ok, but I do not know what that means. Can you help?
Thank you

rrobinett, 2024-06-07:
Is this an rx888 SDR?
If so, is it directly connected to the host?
If so, then I suspect your host is not powerful enough

Rob Robinett
AI6VN
***@***.***
mobile: +1 650 218 8896

HB9VQQ, 2024-06-07:
[image]

rrobinett, 2024-06-07:
Is your HF+ connected through a USB hub?

HB9VQQ, 2024-06-08:
No it's not

rrobinett, 2024-06-08:
At KFS the Airspy HF+ runs well on a Pi4

ka9q, 2024-12-17:
I don't see lost data on the airspyhf+ (or airspy R2) on my systems, mainly Pi4s. I only have the airspyhf+; how is the discovery different?

ka9q, 2025-01-12:
Are you still dropping samples from your airspy? (The fact that you named your machine 'rx888' kinda threw me for a while.)

ka9q, 2025-01-12:
Haven't heard anything in a while, so I'm closing this. If anyone still sees the problem, and your hardware is good (and not overloaded) we can revisit it.

closed 2025-01-12 by ka9q

rrobinett, 2025-01-12:
I have just started a test with an Airspy HF+ feeding a Pi4 and saw the
dropped packets messages when the sample rate was set to the default
912,000.

samprate = 768000

 I no longer see the dropped spots messages, so I think 768000 should be
the default.

Rob

ka9q, 2025-01-12:
Very interesting, thanks. It could be the USB system (it's only using USB2) or the firmware in the Airspy HF+. Hard to see how it could be anything else, so yeah, 768k is probably best as a starting default.

The FFT for 912 kHz at a 50 Hz framerate 1/5 overlap is 228,800 points, which factors to 2 2 2 2 2 2 5 5 11 13. That's not a recommended value for FFTW3 (you're only allowed one factor of either 11 or 13) but even if the FFT were too slow (it isn't) this wouldn't show up as dropped USB samples. 768 kHz uses 19200 points, which is fine:  2 2 2 2 2 2 2 2 3 5 5.