ka9q/ka9q-radio #57: Not an issue

cmh716, 2024-09-30:
My apologies for opening this here, but I didn't see a discussion section. Are there plans to get the rx888 working on uhf and vhf? If not, maybe something like the FoboSDR? I'm thinking it would be nice to have one that covers airband and the entire 2m band. Another could cover all of 70cm, GMRS and a ton of law enforcement frequencies. I suppose multiple hackrfs could work, but that's a lot of Pi's and splitters. Also, fyi radiod works on parallels with rx888 passed to an ARM unubtu vm. Unsupported, but it works. Thanks for sharing your hard work!

rrobinett, 2024-09-30:
For VHF/UHF reception, Ka9q-radio supports the $169 Airspy R2 (
https://v3.airspy.us/product/a-airspy/.
It supports external GPSDO clock input, so I don't think there is any cost
advantage to the RX888 and even when Phil has it working on VHF/UHF we
don't yet know how well it will perform

rob

cmh716, 2024-09-30:
Thanks for the response. To cover the same range with an Airspy.
We'd need 5 of them. Am I mistaken?

rrobinett, 2024-09-30:
Yes, whatever VHF/UHF SDR you choose will require one SDR per band.

One can envision a multi-band downconverter which for example would convert:

50-54.Mhz     => 5-9 Mhz
144-148 MHz => 10-14 Mhz
430-440 Mhz  => 15-25 Mhz
etc...

Those outputs could be mixed and fed to a single Rx888

Perhaps such. device already exists.  If not, perhaps it would cost so much
and/or introduce impairments that make it less desirable than a bank of
four Airspy R2s or similar SDR

cmh716, 2024-09-30:
While expensive, the new FoboSdr could handle 50Mhz, so the entire 70cm band to GMRS. It seems to have better driver support considering it has soapy drivers, etc. of course, it's vastly more expensive than rx888

rrobinett, 2024-09-30:
If you configure KA9Q-radio to run the RX888 at 129.4 MHz sample rate, your
WD site can decode all 2200-6M bands simultaneously.

You will need a good 10+ dB LNA ahead of the RX888 and a 10 MHz shelving
filter (and a good antenna) to achieve reasonable sensitivity on 6m.

To maintain good sensitivity across the whole 0-55 MHz spectrum, many
antenna systems will need to be fed through a 10 Mhz high pass shelving
filter like  Paul WB6CXC's device offered at his Turn Island Systems site:
https://turnislandsystems.com/sdr-front-end-filter/. But that filter
includes a sharp 30 MHz LPF which would need to be bypassed for this 6m
reception.

For 6m WSPR reception you will need to feed the RX888 from a GPSDO using
Paul WB6CXC's adapter kit:
https://tapr.org/product/rx888-clock-kit-and-thermal-pad/

cmh716, 2024-10-01:
Thanks for the information! I've preordered the amplified filter from Turn Island. I suppose I'll figure out my HF setup first and figure out VHF/UHF later. RTLSDR-airband supports soapy drivers, so maybe that could work with higher bandwidth sdr devices for what I want to monitor on uhf/vhf, but I like the design of ka9q-radio because it can do so much more. It's really an impressive project.

cmh716, 2024-12-04:
For anyone wondering, I bought a Fobos SDR to test on VHF/UHF. It has Soapy drivers and it seems to work, but I can only get 20Ms/s without buffer errors when using RTLSDR-Airband. It seems KA9Q-Radio has a more efficient design.

closed 2024-12-16 by cmh716