ka9q/ka9q-radio #58: iface and ttl not being obeyed in [global] section

ke5gdb, 2024-10-02:
I'm working with a Raspberry Pi 5 running the latest version of Pi OS (Debian 12.6 / Bookworm). In my `[global]` section of my radiod@.conf file, I specify:
```
iface = lo
ttl = 0
```
but the multicast traffic is still pushed out through eth0 or wlan0, and never lo. The only way to push it through the loopback interface is if I add a route by means of `sudo ip route add 239.0.0.0/8 dev lo table local`. 

Is this the expected behavior? Should I just add something to the .service file to establish this route before radiod runs?

73!

rrobinett, 2024-10-02:
> (Debian 12.6 / Bookworm). In my [global] section of my ***@***.*** file,

I use ttl = 0 and don't specify any  = xxx, and the traffic does not go out
the physical ethernet port

ke5gdb, 2024-10-03:
That did the trick! Thanks!

73 de KE5GDB

closed 2024-10-03 by ke5gdb