ka9q/ka9q-radio #62: Problems with "powers" utility

argilo, 2024-11-05:
I'm helping with development and testing of radiosonde_auto_rx, and noticed some problems with the "powers" utility:

1. If the specified SSRC does not exist yet, powers creates it, but the first returned row is garbage:
```
$ powers radiosonde.local -f 403025000.0 -w 10000 -b 10 -i 1 -s 1002 -c 2
Tue 05 Nov 2024 17:03:54.946267 UTC, 402974976, 403064976, 10000, 10, -100.0, -100.0, -100.0, -100.0, -100.0, -439.5, -100.0, -100.0, -100.0, -100.0,
Tue 05 Nov 2024 17:03:55.963905 UTC, 402974976, 403064976, 10000, 10, 28.0, 28.3, 28.5, 28.4, 27.8, 27.5, 27.5, 27.3, 27.3, 27.3,
```
2. The first row of data is requested and displayed immediately, regardless of the integration interval. As a result, it may contain data that was integrated over a longer or shorter period than requested.
3. 500ms after each row of data is requested, an output timer fires, which seems to reset the spectrum bin data. This shortens the requested integration time by 500ms.
4. At one point, it seemed like a timer was firing every 500ms and resetting the bin data, thus causing the integration interval to be very short all the time. (I haven't been able to reproduce the conditions that caused this yet.)
5. The time format (`Tue 05 Nov 2024 17:03:54.946267 UTC`) seems to be intended for human consumption. `2024-11-05T17:03:54.946267Z` might be more appropriate.

argilo, 2024-11-05:
I had a peek at the code, and it seems like (4) could occur if `silent` was false for the SPECT_DEMOD channel, which would cause the output timer to restart. I'm not sure how to get it into that state, though.

I also noticed that the code that responds to commands calls `send_radio_status` twice, and each of those calls `encode_radio_status`, which clears `chan->spectrum.bin_data`. Thus the second response (which is probably not used by powers) has all bins set to zero. `chan->status.blocks_since_poll` is not reset until the single `reset_radio_status` call which occurs later. Perhaps the bin data should be cleared there instead?

argilo, 2024-11-05:
After looking at this more closely, it appears (4) may be triggered because radiosonde_auto_rx was accidentally issuing a `tune` command for the SPECT_DEMOD channel.

argilo, 2024-11-07:
I updated the original issue to include one more problem (number 6).

ka9q, 2024-12-17:
I'm considering some important changes for the spectrum power measurement feature, specifically exponential (memoryless) smoothing for the bin energies instead of the current integrate-and-dump-when-read mode. Any comments?

darksidelemm, 2024-12-17:
Unsure exactly what impact this would have on us?
We definitely need to be able to integrate over some time period, so we don't miss pulsing signals (e.g. RS41s). This integration time needs to be at least a few seconds.

ka9q, 2024-12-17:
Right, data would be available immediately but it wouldn't mean anything until the filter has settled down.

ka9q, 2024-12-19:
OK, I propose to make the smoothing time constant a settable parameter. I.e. if you set it to 1 sec, a value will decay to 1/e in 1 sec. I'm going to apply this to all the channel parameters that are currently smoothed or averaged, not just the power spectral data.

argilo, 2025-02-12:
> 1. If the specified SSRC does not exist yet, powers creates it, but the first returned row is garbage

On this particular issue, I see that when powers is compiled in debug mode, it aborts with an assertion failure:

```
powers: powers.c:331: extract_powers: Assertion `l_ccount == l_count' failed.
Aborted (core dumped)
```

I added a `printf` to inspect the values of `l_ccount` and `l_count`, and it appears that `l_count` is invalid:

```
$ ./powers -f 91m0 -w 1000 -b 100 -c 5 -i 1 -s 1243 fm.local
Can't enable multicast option on loopback interface lo
ioctl (set flags): Operation not permitted
l_ccount=100 l_count=65536
powers: powers.c:332: extract_powers: Assertion `l_ccount == l_count' failed.
Aborted (core dumped)
```

argilo, 2025-02-12:
The problem appears to be that radiod doesn't include a `BIN_DATA` field in the packet when no data is available:

https://github.com/ka9q/ka9q-radio/blob/81312bc6e87660918f78f2ab9044e723af9f32a4/radio_status.c#L640-L656

But powers assumes `BIN_DATA` will always be present, and leaves `l_count` uninitialized if it is missing:

https://github.com/ka9q/ka9q-radio/blob/81312bc6e87660918f78f2ab9044e723af9f32a4/powers.c#L308-L317

argilo, 2025-02-13:
I guess an appropriate fix might be:

1. Gracefully handle the no-`BIN_DATA` case in the `extract_powers` function.
2. If no bin data is returned, retry after a short time. (One block duration?)

ka9q, 2025-02-13:
OK, try the patch I just pushed. It retries if the response doesn't contain both BIN_DATA and BIN_COUNT, or if they're inconsistent. I also use parse_frequency on the --freq argument; is it ok to use the heuristic flag (guess scale if the number is unreasonably low and doesn't contain an explicit radix)?

ka9q, 2025-02-13:
I currently replace zero powers with -100 dB, which can be higher than actual possible values. I'm thinking of two fixes: find the lowest actual power in dB and use that same value for zero power, or insert -inf. I suspect the latter might break some code.

ka9q, 2025-02-13:
Bins with exactly zero energy are unlikely except in the pathological case of all zeroes (e.g., no input) but it's worth checking for anyway. I'm pushing over a patch that uses the lowest non-zero energy as the level for any zero power bins.

argilo, 2025-02-13:
The recent changes mostly look good. But it seems like a short sleep is needed in the case where powers receives an invalid response. Currently it sends a new command immediately, and radiod fails to respond, so powers has to wait for a timeout (default = 1 second).

Adding another one of these seems to fix it:
```diff
     int npower = extract_powers(powers,sizeof(powers) / sizeof (powers[0]), &time,&r_freq,&r_bin_bw,Ssrc,buffer+1,length-1);
     if(npower <= 0){
       fprintf(stderr,"Invalid response, length %d\n",npower);
+      usleep(10000); // rate limit, just in case
       continue; // Invalid for some reason; retry
     }
     // Note from VK5QI:
```

argilo, 2025-02-13:
On that topic, the requirement for clients to rate-limit their command is a bit of a pain. Is that something that could be fixed in radiod, or does the requirement come from somewhere else?

argilo, 2025-02-13:
> is it ok to use the heuristic flag (guess scale if the number is unreasonably low and doesn't contain an explicit radix)?

I think that will be fine for Auto-RX. It passes in the frequency in Hz, which will be over 400000000.

ka9q, 2025-02-13:
Add a decimal point and it will be unambiguous.

argilo, 2025-02-13:
Would that approach run into locale issues though?

ka9q, 2025-02-13:
Rate limiting just seems like a good idea in general, to keep some bug or misconfiguration from spewing out back-to-back packets at line rate. Also it's not really meaningful to query radiod about some channel since the data is only updated once per block time (20 ms).

argilo, 2025-02-13:
> Would that approach run into locale issues though?

I remembered that it would not, because powers does not use `setlocale`.

argilo, 2025-02-13:
> Rate limiting just seems like a good idea in general

I agree that it's a good idea. But I expect radiod would still need to be able to respond to commands arriving at any time, since they may come from multiple clients.

ka9q, 2025-02-13:
Good point. Although the locale may not be a problem with 'powers', it could be for other programs that do use it, so I just added 'h' (Hertz) to the set of characters that may be used in frequency strings. The ITU uses it to mean unity scaling, e.g., 100h0 means 100.0 hertz.

ka9q, 2025-02-13:
The radio_status thread accepts commands at any time and passes them to the channel, but they won't get executed until the beginning of the next block. (Execution used to be asynchronous but I did this in part to avoid some thorny race conditions without extensive locking.) At the moment only one per channel can be pending; additional ones will be ignored. I suppose this could be a FIFO queue, but commands are (intended to be) idempotent, the usual command is a poll, and it doesn't make sense to do more than one poll per block time. Because everything is multicast, I'm leery about unnecessary traffic. But I'm certainly open to suggestions.

ka9q, 2025-02-13:
I'm inserting 10 millisecond sleeps after errors. This seems like a good balance between rate limiting and responsiveness, since 10 ms is half the usual radiod block time.

ka9q, 2025-02-13:
By the way, I'm open to suggestions on how the entire spectrum analyzer thing works. I considered emitting spectrum data as an continuous output stream, but I chose the current method so the user could set the integration time by simply changing the polling rate. That got complicated when I started to send periodic status on the data channel, though I suppose I could reset the integrator state only in response to a direct poll.

Also, the spectrum data responses are usually large enough to require IP fragmentation, which is usually best avoided. It works OK here because our multicasts are limited to the LAN where packet losses are hopefully rare. Even with fragmentation there's an upper limit of 64 KB on the complete message.

argilo, 2025-02-13:
The latest changes to powers are working correctly here. When creating a new channel, one "Invalid response, length 0" occurs, and powers quickly recovers from that. No errors appear when using an existing channel.

The changes are also working well with Auto-RX.

argilo, 2025-02-13:
Can we leave this open for now? There are a few other issues with powers which I don't believe have been addressed yet. I may take a look at those at some point.

ka9q, 2025-02-13:
I'm running my latest with both ka9q-web and auto_rx and it seems to work fine so far.

ka9q, 2025-02-15:
I'm redoing spectrum.c. It currently uses a dummy output filter to produce the selected segment of frequency bins but that's rather wasteful. I'm rewriting it to directly use the proper segment of the master's frequency array. It's working with the rx888, but I have to properly handle three different input configurations: complex input (e.g., airspyhf); real input with the spectrum right side up (e.g., rx888); real input with spectrum inverted (e.g., airspy r2). I need to test further with all three before I release the changes.

argilo, 2025-02-16:
That sounds fine. I presume these changes are internal, and shouldn't have any effect on powers' output.

ka9q, 2025-02-23:
No more comments here lately, so I'm closing this.

closed 2025-02-23 by ka9q