ka9q/ka9q-radio #66: Byteswap default in pcmcat changed

argilo, 2024-11-27:
After a recent change to pcmcat (made in 5e606d5f9dd0fd9f3881f1f736d6bf7f0d6434b5), Auto-RX stopped working. It looks like the reason is that pcmcat previously defaulted to byteswapping, but now only does so when the stream type is 10 or 11 (44100 sps, S16BE). Auto-RX uses type 113 (48000 sms, S16BE).

Adding the `-b 1` argument works around the problem, but I'm not sure whether it was intentional to change the default behaviour for stream type 113 (and the other S16BE stream types).

argilo, 2024-11-27:
I have proposed to explicitly request byteswapping in Auto-RX with the `-b 1` argument: https://github.com/projecthorus/radiosonde_auto_rx/pull/947

ka9q, 2024-12-19:
The new pcmrecord automatically swaps S16BE to be S16LE, so it always writes the latter. It also recognizes float32, float16 and Opus, as well as the sample rate and channel count and writes the correct formats. Hopefully this solves your problem so I'm closing this.

argilo, 2024-12-19:
Yeah, I see there's now an option to output to stdout in pcmrecord, so I expect that Auto-RX can switch to that. For the time being, we're using pcmcat with `-b 1`.

ka9q, 2024-12-19:
Yeah, I'm thinking of doing away entirely with pcmcat and pcmspawn, unless that would break too much. Their functions ought to be completely superseded with --stdout and --exec. Note: a wav header is always added to pcm output as there is no other easy way to get metadata to the client. You can probably just ignore it if you want to treat it as raw and can tolerate the initial noise burst.

argilo, 2024-12-19:
> You can probably just ignore it if you want to treat it as raw and can tolerate the initial noise burst.

Yeah, I expect Auto-RX will tolerate it.

Perhaps it could be made optional (via a command line argument) to save downstream applications from having to strip it off if they don't want it.

ka9q, 2024-12-19:
I've been thinking about doing just that, adding a --raw option.

ka9q, 2024-12-20:
If you're happy with my plan to replace pcmcat with the new pcmrecord, I will probably create a short shell script named "pcmcat" that will invoke "pcmrecord" with the --stdout flag and pass along other flags. Any I should add besides maybe --raw before I close this issue?