ka9q/ka9q-radio #67: Compilation issue after Commit b2270dc

bminish, 2024-11-28:
the most recent change ( Commit b2270dc ) dated 27/11/2024  in the Makefiles broke compilation on my ubuntu 24.04 setup.

Presumably it's down to a new library requirement but I am not able to ascertain what library I need to add 

For now on line 121 of the Makefile

$(CC) -g -o $@ $^ -lopus -logg -lbsd -lm -lpthread

removing -logg allows for compilation. 

73
EI6IZ

bminish, 2024-11-29:
the ubuntu 24.04 machine is a minimal install of ubuntu 24.04 with only the libraries required to build ka9q-radio and it's running on x86_64 

The linux makefile built ok on my pi5 with a minimal raspbian install

argilo, 2024-12-02:
You'll need to install the `libogg-dev` package.

The install documentation was recently updated to include it: 12220bc605a1e9073182b787f154042e645c50e4

closed 2024-12-04 by bminish