ka9q/ka9q-radio #79: Fobos SDR Help

cmh716, 2024-12-18:
Sorry for opening an issue—I am not aware of any forum where I can ask this question. I am trying to add support for FobosSDR, and admittedly, I don't fully understand what I'm doing, but I am trying to learn. This is my first attempt at writing anything related to SDRs. I'm sure I have a lot wrong, but would anyone be willing to help me get this working?

Code: https://github.com/ka9q/ka9q-radio/compare/main...cmh716:ka9q-radio:fobos

It runs, and returns samples from the SDR in rx_callback, but when I view the frequency from "control" it looks like nothing is being processed.

This SDR does not have AGC; right now, I am focused on RF input only for VHF/UHF (not direct sampling).  The ADC can do up to 80Ms.

libfobos spec is here: https://github.com/rigexpert/libfobos/blob/main/fobos/fobos.h
Sample program here: https://github.com/rigexpert/libfobos/blob/main/eval/fobos_recorder_main.c

Maybe one of the experts on here can point me in the right direction?  Thanks!

cmh716, 2024-12-19:
I seem to have it somewhat working now. I can record using pcmrecord, but it sounds distorted (but I can make it out).

ka9q, 2024-12-19:
Do you want me to merge this in? I will if you promise to support it...

cmh716, 2024-12-19:
Hi Phil,
Its not 100% working yet. For some reason, recorded audio using pcmrecord doesn't sound right. Once I figure it out and ckean it up,  I’ll submit a pull request and sure, ill gladly support it.

ka9q, 2024-12-19:
Check the byte ordering. That's a common mistake.

cmh716, 2024-12-19:
Thank you.  I have this working now.  I will clean up my code and submit a pull request within the next few days.  But I have one question Phil.

I noticed the recent SDRPlay implementation, they created a Makefile.sdrplay, so it's not part of the default installation and not built in the CI pipeline (i think).  Do you want the same for Fobos?  I notice one challenge with the fobos -- there are not yet libfobos drivers in the package manager so I must build from source.  I think I can add Fobos to the main Makefile.linux, but I would need to incorporate compiling libfobos from source into the Makefile and also the CI pipeline.

Alternatively, I can maintain a separate Makefile.fobos until there is a proper libfobos package out there and write documentation on how to get Fobos SDR going.

ka9q, 2024-12-19:
I haven't mainlined SDRPlay because there's no open source driver. I'm willing to mainline your stuff as long as the full source is included and all the tools needed to build it are available in package distributions. But let's keep it separate for now. When you get some operational experience, I can merge it.

cmh716, 2025-01-08:
Addressed in https://github.com/ka9q/ka9q-radio/pull/92

closed 2025-01-08 by cmh716