ka9q/ka9q-radio #81: Strange issue with Airspy Mini

MikeTango, 2024-12-19:
Hi,

when installing the ka9q-radio for auto_rx, i noticed a trough at the very end of the spectrum. 
[image]

I then set the highest frequency to 404 mhz and it displays this behaviour in the spectrum of auto_rx: 
[image]

This picture is without any preamp or antenna attached. When changing the highest frequency in the config to 1mhz higher: 
[image]

any advice?

ka9q, 2024-12-20:
I can't see the scales on the plot. What is the center frequency, sampling rate, spectrum scale, etc? Offhand it looks like you're seeing the rolloff region between the top of the output spectrum where the high pass filter begins to roll off and the local oscillator (LO), which is the spike in the middle of the trough. Above that you see the mirror spectrum.

The Airspy is essentially an SSB receiver that produces a wide lower sideband from about -600 kHz to about -9.4 MHz, sampled at 20 MHz by a single A/D converter. So if you try to look above the LO, you'll just see the mirror image of the lower sideband. ka9q-radio knows to use only the proper passband region, so it will automatically place the LO depending on the requested channel frequencies (unless the LO is manually set with the 'frequency' entry in the [hardware] section). E.g., if you want the radiosonde band between 400-406 MHz, the LO should be in the vicinity of 407 MHz.

MikeTango, 2024-12-20:
Thanks for the quick reply, that absolutely explains the behaviour, i kind of had an inkling that i had seen something like that before. The spike moved with the set frequency. Since the Airspy mini is only good for a portion of the band i will have to acommodate, auto_rx darksidelemm recommended .6mhz offset to not waste anything but it looks like she's running at full choo choo now. 

[image]

This is orders of magnitude better than spyserver.

MikeTango, 2024-12-20:
[image]

closed 2024-12-20 by MikeTango