ka9q/ka9q-radio #82: Strange, but my ini parser gives an error

AndyLinwood, 2024-12-26:
Strange, but my ini parser gives an error. When starting the radiod

`./radiod radiod@sdrplay.conf`

I get an error

`Loading config file radiod@sdrplay.conf...
iniparser: cannot open radiod@sdrplay.conf
Can't load config file radiod@sdrplay.conf`

The configuration file is located /etc/radio/

argilo, 2024-12-29:
Unless you specify the full path, radiod will look in the current directory for the configuration file. Try `./radiod /etc/radio/radiod@sdrplay.conf`.

ka9q, 2025-01-12:
Right. _In addition to_ reading this file, I now also look for a directory with the same name with ".conf.d" appended. If it's present, it will sort all files in it that end in ".conf" and append them to the contents of the main .conf file. This is optional, but it lets you break config files into more manageable pieces.

ka9q, 2025-01-12:
There doesn't seem to be a bug here, so I'm closing this.

closed 2025-01-12 by ka9q