ka9q/ka9q-radio #101: Issues when working with auto_rx

MikeTango, 2025-01-30:
Hi, just want to ask here whether this 

https://github.com/projecthorus/radiosonde_auto_rx/issues/975

is caused by ka9q-radio, auto_rx...or me.

regards,

MT

ka9q, 2025-01-30:
I read the thread, but I'm not sure what the problem is. You can set the LO frequency in the Airspy to anything you want, limited only by the precision of the fractional-N synthesizer. This is on the order of hertz, not megahertz. ka9q-radio will then use the _actual_ frequency of the LO to set the second LO (the one in software) to get the desired dial frequency. (A pet peeve of mine regarding drivers for these devices is that they pretend to tune in, e.g, 1 Hz steps while actually tuning in much larger steps that are rarely if ever multiples of 1 Hz _and then not telling you the actual frequency_. So I have to find how the device firmware calculates the synthesizer settings, and then "shadow" them to find the actual tuned frequency.

Are you perhaps talking about what happens when you don't set the frequency in the config file but let me do it when you tune or retune a channel? In general I try to move it by the least amount that will bring the desired passband just within the range of the tuner. ka9q-radio knows the edges of the front end's filter skirts and takes them into account along with the desired channel bandwidth. But this can be error prone especially if you are interactively tuning a receiver; entering a decimal point incorrectly can take the tuner off into the weeds, suspending all your other channels. So for a dedicated application like receiving sondes in a well defined band, I recommend locking the front end at an appropriate spot, e.g., 406.6 MHz or maybe a little higher. The high end of the receiver filter (ie., the low end of the covered range) is about 9.4 MHz below the carrier, so you go all the way up to 409.4 MHz and still get everything down to 400 MHz.

ka9q, 2025-02-05:
Is there still an issue here?

argilo, 2025-02-06:
My understanding is that @MikeTango is explicitly setting the hardware frequency with `frequency =` but the value is being rounded down to the nearest MHz.

I tried to reproduce this with an Airspy Mini, but the issue does not occur on my system. @MikeTango What exact commit of ka9q-radio are you running? What operating system and version are you running? Are you using the Airspy drivers provided by your distribution?

MikeTango, 2025-02-06:
Hi, 

that is apparently the case. No matter what decimal i put into the config, it only jumps in full mhz steps.

 The software running on top is the radiosonde_auto_rx stack which also provides the spectrum graphics. I am running raspbian stock and had to choose a commit as detailed in https://github.com/projecthorus/radiosonde_auto_rx/wiki/KA9Q%E2%80%90Radio-Setup-Notes , but that might have changed in the meantime. The wiki mentions libairspy etc, so i am assuming i'm running stock drivers supplied by raspbian. 

I currently have no access to the machine running it and may only get to it on the weekend to pull the actual commit / version data.

argilo, 2025-02-06:
Actually, I won't need that additional information. I was able to reproduce the issue on my Raspberry Pi by changing the system locale to `de_DE.UTF-8`.

@ka9q I suspect the bug is that main.c calls `setlocale`, which changes the behaviour of `strtod`, which is used by `parse_frequency` in misc.c.

MikeTango, 2025-02-07:
Good morning from indeed Germany, could that possibly be because german uses , comma as a decimal and not . ?

ka9q, 2025-02-07:
Oh boy, this could be a problem. 50 years of carefully counting digits in large numbers has almost driven me insane, so I regularly use the ' (tick) option in printf format strings to add group separators. 'control' relies on these to get the proper format in frequency displays. But the tick option works only if you have the locale set, which is why I call setlocale in everything, passing the $LANG environment variable. So I don't want to remove setlocale(). But I certainly will try to fix anything that gets broken in locales that don't use the US convention for separators and decimals.

ka9q, 2025-02-07:
OK, I just pushed over a change to parse_frequency() in misc.c that uses localeconv() to determine the local decimal point. I tried it with the German locale in my 'control' program and got the expected results so it should also work correctly when parsing frequencies in the config files. Let me know of any more of these nonportabilities, it's easy to miss them when you're American.

MikeTango, 2025-02-07:
At least you're using SI units

ka9q, 2025-02-08:
You guys got me looking at other locale-related stuff. On a hunch I tried fr_FR.UTF-8 and found that although frequencies were correctly rendered as (eg) 1 000 000,000 they were not properly positioned in the 'control' display. The problem turned out to be that the French grouping character is the Unicode non-breaking-space, which in UTF-8 is 3 bytes long. strlen() counted this as 3 bytes. So I added a strlen_utf8() function that did it correctly.

Please let me know of any other locale-related bugs; since we Americans originally invented computers, we tend not to notice problems in handling other languages. :-)

closed 2025-02-08 by ka9q

MikeTango, 2025-02-09:
> since we Americans originally invented computers

Hate to break it to you but the computer as we know it was invented by German Konrad Zuse and predates the ENIAC by 5 years

https://en.wikipedia.org/wiki/Z3_(computer)