ka9q/ka9q-radio #105: ncursesw not present on macOS

argilo, 2025-02-06:
The macOS CI build began failing with 118e8465dccef0218b4c793a6ef7be3f82463570, which replaced ncurses with ncursesw in Makefile.osx. It appears that ncursesw is not supported on macOS: https://invisible-island.net/ncurses/ncurses.faq.html#what_platforms

ka9q, 2025-02-07:
Works for me. I use mac 'ports' and have the ncurses (no 'w') package installed. Yet 'port provides /opt/local/lib/libncursesw.a' returns 'ncurses' so apparently that package provides both ncurses and ncursesw.

ka9q, 2025-02-07:
My primary desktop (laptop, actually) is macos, so I am often running 'control' and 'monitor' on it. Both uses curses. Going to ncursesw fixed the screwed-up boxes I had been getting with ncurses.

argilo, 2025-02-07:
> I use mac 'ports'

OK, that explains why it works for you and not in CI, where Homebrew is used to install the dependencies. I'll have to check whether it's possible to get ncursesw through Homebrew.

argilo, 2025-02-07:
The issue was that CI was building against the system-provided ncurses library, instead of the one provided by Homebrew. I fixed that in #106.

PR #106 (2025-02-08): Link to Hombrew's ncurses in CI
Fixes #105.

Homebrew's ncurses is a so-called "keg-only" package, which does not symlink into the usual locations to avoid overriding the system-provided ncurses package. Thus, we need to opt into using it by providing the correct paths to the compiler, which I've done here.
closed 2025-02-08 by ka9q