ka9q/ka9q-radio #108: Is ft8-decoded broken?

cmh716, 2025-02-08:
Hi All,

I am setting up ft8-decoded for the first time and testing it via:

/usr/local/bin/jt-decoded -v -d /home/chris/ft8 -x /usr/bin/jt9 -8 ft8.local

I get the following errors:

`Error termination. Backtrace:
At line 120 of file /home/uwe/jt/wsjtx/lib/jt9.f90
Fortran runtime error: Bad integer for item 1 in list input

Error termination. Backtrace:
#0  0x7e9936918e59 in ???
#1  0x7e9936919a71 in ???
#2  0x7e9936ba5e91 in ???
#3  0x7e9936ba91ca in ???
#4  0x7e9936baa02f in ???
#5  0x560d3e2d4f79 in ???
#6  0x560d3e2d483e in ???
#7  0x7e99362271c9 in ???
#8  0x7e993622728a in __libc_start_main
#9  0x560d3e2d48bd in ???
#10  0xffffffffffffffff in ???
#0  0x794b5ca33e59 in ???
#1  0x794b5ca34a71 in ???
#2  0x794b5ccc0e91 in ???
#3  0x794b5ccc41ca in ???
#4  0x794b5ccc502f in ???
#5  0x58c3c882ef79 in ???
#6  0x58c3c882e83e in ???
#7  0x794b5c3421c9 in ???
#8  0x794b5c34228a in __libc_start_main
#9  0x58c3c882e8bd in ???
#10  0xffffffffffffffff in ???
#0  0x78897b657e59 in ???
#1  0x78897b658a71 in ???
#2  0x78897b8e4e91 in ???
#3  0x78897b8e81ca in ???
#4  0x78897b8e902f in ???
#5  0x6248bcad6f79 in ???
#6  0x6248bcad683e in ???
#7  0x78897af661c9 in ???
#8  0x78897af6628a in __libc_start_main
#9  0x6248bcad68bd in ???
#10  0xffffffffffffffff in ???
At line 120 of file /home/uwe/jt/wsjtx/lib/jt9.f90
Fortran runtime error: Bad integer for item 1 in list input`

I have tried a few different versions of wsjtx, but I don't think that is the issue.  jt-decoded is successfully recording to /home/chris/ft8, because when I run find /home/chris/ft8 -name *.wav -exec  /usr/bin/jt9 -8 {} \; it successfully decodes FT8 from the recorded files.  Maybe I am doing something wrong?  The behavior is the same via systemctl recording to /run/ft8

cmh716, 2025-02-08:
Or is everyone using pcmrecord for ft8/ft4 and WSPR now?  If so, I'm happy to use that instead

rrobinett, 2025-02-08:
Wsprdaemon installs and configures those itself
So if you want to do that yourself, then disable KA9Q management in WD.conf

Rob Robinett
AI6VN
***@***.***
mobile: +1 650 218 8896

cmh716, 2025-02-08:
Thank you Rob.  I hadn't yet considered using Wsprdaemon, and was looking at only kaq9-radio so far, but if that is the strategy, I can look at it.

cmh716, 2025-02-08:
unrelated to this issue, I see a lot of references to modes.conf, but I cannot find it on my system or in this repo.  Is this demised?

scottnewell, 2025-02-08:
I think modes.conf is now share/presets.conf: 

main.c:dictionary *Preset_table;   // Table of presets, usually in /usr/local/share/ka9q-radio/modes.conf or presets.conf

ka9q, 2025-02-08:
Yes, I renamed modes.conf to presets.conf some time ago, but I tried to maintain backward compatibility for the old name. "Presets" is much more descriptive of what the file actually contains.

ka9q, 2025-02-08:
I could fix it, but I plan to deprecate jt-decoded pretty soon.

cmh716, 2025-02-08:
No, that’s fine, Phil. I was just testing. I will close. Thanks for the info on the presets.conf

closed 2025-02-08 by cmh716

rrobinett, 2025-02-08:
What are your plans on replacing jt-decoded?
I have been asked to add code to WD to filter out spots with the same
callsign as the WD Reporter_ID.  Wsprnet drops those spots anyways, and
since WD feeds the same spots to PSKReporter it too will not get those
spots.

But since the FT8/8 spots come from jt-decoded and go directly to the
psk_uploader, I will have to modify one of those in order to suppress those
FT8/4 spots.

So I hesitate to modify those if you are soon going to deprecate them.

waterwin, 2025-02-21:
Along the same lines my question. I am setting up a Fobos SDR for ka9q-radio. This works well for RF (50 MHz - 6000 MHz) and for HF Direct Sampling the 100kHz-25 MHz range. I activated the FT8 decoder and in its log it can be seen that it is working great. But the spots are not uploading. I do not find any code references to pskreporter and only in this issue a reference to psk_uploader. What do I need to install or where can I find the needed upload scripts created by Phil Gladstone for this task. My other setup, RX888-wsprdaemon-radiod is uploading to pskreporter without issues, this runs (obviously) from a different computer.