ka9q/ka9q-radio #109: Excessive WFM deemphasis

argilo, 2025-02-10:
WFM audio output is very muffled. I dug into it, and found that the problem is in the deemphasis time constant calculation, which uses the composite sample rate (384000) instead of the audio sample rate (48000):

https://github.com/ka9q/ka9q-radio/blob/5d51cef5e0ccf7a16ea320bda825e93a0d03ec02/modes.c#L289-L295

I don't understand the code well enough to know what the correct fix is, but I did observe that replacing `chan->output.samprate` with a hard-coded 48000 fixed the audio output.

argilo, 2025-02-10:
Perhaps this would be an appropriate fix?

```diff
diff --git a/modes.c b/modes.c
index 0cd5f5a..6af19ab 100644
--- a/modes.c
+++ b/modes.c
@@ -290,7 +290,8 @@ int loadpreset(struct channel *chan,dictionary const *table,char const *sname){
     char const *cp = config_getstring(table,sname,"deemph-tc",NULL);
     if(cp){
       float const tc = strtof(cp,NULL) * 1e-6;
-      chan->fm.rate = -expm1f(-1.0f / (tc * chan->output.samprate));
+      unsigned int samprate = (chan->demod_type == WFM_DEMOD) ? 48000 : chan->output.samprate;
+      chan->fm.rate = -expm1f(-1.0f / (tc * samprate));
     }
   }
   {
```

ka9q, 2025-02-11:
Good one! I had noticed it sounding a little muffled but I thought it was just my hearing.

The WFM demod is a bit of a kludge because I don't have a general way to represent collections of filters running at different sample rates. (Note the filtering in the control display is a little misleading). It already bends my attempt at keeping ka9q-radio modular (since I already had a standalone stereod) but I have to admit it's a lot easier with it built into radiod. Then again, I shouldn't try to recreate GNUradio either.

closed 2025-02-11 by ka9q