ka9q/ka9q-radio #111: Undefined behaviour in main.c

argilo, 2025-02-11:
ASAN shows that the `pthread_join` call at the end of this code section is sometimes called on a thread that was not created, thereby invoking undefined behaviour:

https://github.com/ka9q/ka9q-radio/blob/14ac7cfe7d626a97b276e6b7232a733c8847a005/main.c#L555-L580

The root cause is that the thread-creating loop skips some sections in the configuration file, but the thread-joining loop does not skip them and attempts to `pthread_join` a non-existent thread.

It looks like an attempt to avoid crashes was made by zeroing `startup_threads`, but this does not remove the undefined behaviour. The thread-joining loop should only join threads that were actually created.

ka9q, 2025-02-11:
That's right, I spent some time trying to find a portable way to tell if a thread pointer is valid, and there doesn't seem to be one. Hence that assert. I'll try to clean this up.

argilo, 2025-02-11:
Maybe just use threads in sequence, with a separate counter that increments whenever a thread is used?

ka9q, 2025-02-11:
That's exactly what I just did. Patch pushed.

ka9q, 2025-02-11:
Thanks for running these tools. With enough eyes all bugs are shallow.

closed 2025-02-11 by ka9q

argilo, 2025-02-11:
Looks good. With that fix in place, I would expect the memset and zero check are unnecessary.