ka9q/ka9q-radio #118: Race condition when setting frontend frequency on startup

argilo, 2025-02-12:
It appears that the new multi-threaded startup has caused a problem setting the frontend frequency on startup, in the case where the frontend frequency is not specified with `frequency=`. The chosen frontend frequency is high enough to accomodate frequencies specified in some demodulator sections, but not all.

I believe the problem is a race condition: all threads start executing simultaneously and read `Frontend.frequency=0`, then calculate a new LO frequency and call `set_first_LO`. Whichever thread does this last is the winner, and it overwrites all the other threads' LO values.

It seems like the read and subsequent write (if any) of `Frontend.frequency` would need to occur inside a mutex to avoid this problem.

argilo, 2025-02-12:
It is the `set_freq` function in radio.c which is subject to races.

argilo, 2025-02-12:
When I saw this come up, I had several sections defined, each containing only a single frequency. That's probably a worst-case scenario for the race condition to occur, in case you need a way to reproduce.

ka9q, 2025-02-12:
Yeah, I knew that multi-threaded startup seemed too easy. I'll think about it. But in general there is no solution when the front end can't satisfy all of the channels, and I'm not really sure anything is better than just letting them contend randomly. Before the multi-threaded startup it was last comer wins.

ka9q, 2025-02-12:
OK, I've pushed over a patch to protect set_freq() with a mutex. In principle all the channels will detect a change in front end tuning and automatically compensate, but I can see how a race might cause one channel to tune the front end off into the weeds where none of the channels can use it.

I think I'm spoiled by direct sampling front ends.

closed 2025-02-12 by ka9q

argilo, 2025-02-12:
> there is no solution when the front end can't satisfy all of the channels,

To be clear, in the issue I described here that was not the case: all frequencies in my config file were close enough that they could be satisfied simultaneously. But the race condition results in the frontend being tuned too low, so that only one channel is satisfied.

argilo, 2025-02-12:
E.g. if the specified frequencies were 100, 101, 102 MHz them the Airspy might be tuned to 100.6 or 101.6 instead of the correct 102.6, depending on the luck of the draw.

argilo, 2025-02-12:
The new mutex does look like it should solve that problem correctly.