ka9q/ka9q-radio #122: Some further information on the Fobos SDR / question

waterwin, 2025-02-13:
LNA Gain has 4 steps from 0-3 
PGA Gain (as some call it) or VGA gain 0-15 
In your generic fobos conf file comments you have these 1 time ok and 1 time vice versa.

cmh716, 2025-02-13:
I have some confusion about the real VGA gain.

The H file says 0..15, but this shows it 0-31:
https://github.com/rigexpert/libfobos/blob/main/fobos/fobos.c#L1431

Their newer "agile" library says 0..15:
https://github.com/rigexpert/libfobos-sdr-agile/blob/main/fobos/fobos_sdr.c#L738

To your point, we must fix the commented options to swap them, at minimum. Ultimately, I wasn't sure whether to use 0..15 or 0..31, so I chose 0..31.

waterwin, 2025-02-13:
I 'assume' that the 0-15 is the number of steps and the change is 2 dB per step 0-31 dB but not sure.
ErwinPE3ES / F4VTQ

 
  

https://github.com/rigexpert/libfobos/blob/main/fobos/fobos.c#L1431

https://github.com/rigexpert/libfobos-sdr-agile/blob/main/fobos/fobos_sdr.c#L738

https://github.com/rigexpert/libfobos/blob/main/fobos/fobos.c#L1431

https://github.com/rigexpert/libfobos-sdr-agile/blob/main/fobos/fobos_sdr.c#L738

   

|  | Virus-free.www.avast.com |

ka9q, 2025-02-13:
These gain settings control the MAX2830 transceiver despite the notation "VGA" on the SKY65405 just ahead of it (the gain of which appears to be fixed at 14 dB according to its data sheet). So I disregarded comments in their library source and went right to the MAX2830 data sheet.

Page 21 in the MAX2830 data sheet says that the high two bits of the gain control register set the gain range:

11 => max gain
10 => medium gain (-16 dB)
0X => low gain (-33 dB)

The same page says that the low five bits control gain in 2 dB steps, with all 1's being maximum gain and all 0's being minimum gain, i.e., -64 dB relative to maximum. Although the max2830 spec sheet doesn't use this terminology, I assume that fobos' "lna_gain" refers to the 2-bit gain range and "vga_gain" to the 5-bit 2dB gain step, and I compute the overall relative gain with

frontend->rf_gain = 2 * sdr->vga_gain + (sdr->lna_gain == 2 ? 16.0 : sdr->lna_gain == 3 ? 33.0 : 0);

I wrote the function fobos_gain() to compute the "vga" and "lna" settings from a desired gain setting, with minimum gain defined as 0 dB and maximum +96 dB. Because the noise figure is best at higher "lna" gain (ie, gain range, see page 11), I use the largest value that is equal to or less than the requested gain. Then I make up the difference with the "vga" steps. 

This seems correct though I haven't tested it extensively. I will probably also implement a software AGC similar to those I've done for the RTL-SDR, Funcube, Airspy R2 and RX888. 

Note that to simplify signal measurements and avoid upsetting the per-channel AGC in the linear demodulator, radiod applies a digital loss equal to the analog gain to maintain constant gain from antenna input to digital output should remain constant regardless of the analog front end gain setting _**IF**_ the analog gains are accurate.

closed 2025-02-13 by ka9q

waterwin, 2025-02-15:
As can be seen from the GNU Source block of RigExpert for the Fobos, the LNA gain has 3 steps : 0-15-30 dB
The VGA Gain has 2 dB steps from 0 -- 30 dB (15 steps)

ka9q, 2025-02-15:
Hm. That's not quite consistent with the information I got from the spec sheet. I haven't tried actually measuring the steps yet.

cmh716, 2025-02-15:
SDRPlusPlus is doing it the way we are (0-31)

https://github.com/AlexandreRouma/SDRPlusPlus/blob/master/source_modules/fobossdr_source/src/main.cpp

This also includes the version of SDRPlusPlus that Rigexpert forked and released. I think this aligns to the data sheet as Phil points out. If Rigexpert has a reason to only use 0-15, they should probably explain why. Maybe we can raise a case on the libfobos GitHub for clarification. I have another case open there for firmware upgrades and they're slow to respond. Of course, their office just got bombed, so they have more important issues right now, but will eventually respond.

waterwin, 2025-02-15:
Chris, I said : VGA Gain has 2 dB steps from 0 -- 30 dB (15 steps), RigExpert is using 2 dB steps, meaning steps of 2 dB each for 0-15 ending at 30 dB.  Did you have the same interpretation of 0-15 ? not being a dB scale but a steps scale ? If not, I am confused ;-)

waterwin, 2025-02-17:
Text in fobos generic is wrong.
vga_gain 0-31; 0 to +62 dB in 2 dB steps (disregarded in direct sample mode)
This has 0-15 steps for a total of 30 dB gain.
Maximum gain including lna gain is 33 + 30 = 63 dB, the text on lna gain is correct
See also the comments in the fobos source for their gnu source block implementation for instance

cmh716, 2025-02-17:
Phil, is fobos_gain called in the updated fobos.c.  Also, I think this is what waterwin is referring to: https://github.com/rigexpert/gr-rigexpert/blob/main/grc/RigExpert_fobos_sdr.block.yml#L45

cmh716, 2025-09-01:
FYI - The latest Fobos firmware is supposed to support 0-31

ka9q, 2025-09-02:
I lost track of this conversation. What would you like me to do? BTW, I now have a signal generator (calibration questionable) that I could use to get a rough idea of the gain settings. In HF direct sampling mode it's already pretty close to 0 dBm -> 0 dBFS (only 0.8 dB off) but that bypasses almost all of the receiver. I just pushed over a patch for this. I've also been using it for VHF airband monitoring but I haven't checked the gain there.

cmh716, 2025-09-02:
Hi Phil,

I think there was a bit of confusion over the VGA gain being 0..15 or 0..31, but it looks like they recently released a new firmware for the fobos and a new version of libfobos and it's 0..31. 

I upgraded my firmware and oddly it has an August 2025 firmware date, but it doesn't say 2.4.0. Also, does the fobos_gain ever get called anywhere?

--- 
https://github.com/rigexpert/libfobos/blob/main/versions.txt

v.2.4.0 
- DC filter improved
- VGA gain fixed, whole range #0..31 is available

cmh716, 2025-09-02:
Also, nice job on the raster. I tried the airband .conf you created and my poor intel 155h struggles with the 700+ simultaneous frequencies, but it can do it at 25Ms. At 50, it stutters.

ka9q, 2025-09-02:
How do I upgrade the firmware? I'm always a little reluctant to do these things out of fear I'll brick my unit. Ordinarily I'd say "please don't tell me I need Windows" but I actually have it around this time if I need it.

ka9q, 2025-09-02:
I guess fobos_gain() is indeed not called anywhere, I could turn it off. I
do call fobos_rx_set_lna_gain() and fobos_rx_set_vga_gain() individually,
which is probably why I don't use the combined function.

Setting gains properly is tricky on SDRs with multiple gain stages, you
really need gain tables that give the optimum setting for each stage for a
desired overall gain. Often there's more than one table, one optimized for
noise figure and the other optimized for intermod distortion. E.g., the
Airspy has this. The Fobos has an entirely different front end so it needs
its own gain tables.

cmh716, 2025-09-02:
I mainly did it to try the higher vga gain but also so subsequent firmware upgrades won't require me to open the case. The funny thing is that after performing the upgrade, I notice my SNR suffers with the VGA on at all so I have it off anyhow. I'd say don't bother.