ka9q/ka9q-radio #124: WFM mono fallback not working

argilo, 2025-02-14:
The following code is intended to switch WFM output to a single channel if the stereo pilot is not present:

https://github.com/ka9q/ka9q-radio/blob/778490c9e5591bb627cc278259900138df8104e5/wfm.c#L248-L253

But it doesn't actually work; monitor still sees two channels, and the audio is garbled.

Perhaps it would be better to respect the `channels` option if the user provides it? I could imagine that we might want to flip between mono and stereo depending on the signal strength, and I expect that changing the number of channels on the fly could be a problem for clients.

ka9q, 2025-02-15:
I have one local station that usually runs without a stereo pilot (KPBS-FM). When I tune to it, I hear mono (and see 1 channel). When I flip to stereo, nothing changes -- but only when I use Opus encoding on the output. With PCM, I hear the garbling you're talking about. Never heard it before so I must have introduced it recently.

On stations with pilots, I can flip between mono and stereo and everything behaves as expected. Because of this, I suspect my pilot detection is falsing. I'll work on it.

argilo, 2025-02-15:
Yes, I'm using PCM. I suspect that pilot detection is working as intended, but wfm.c is producing 1 channel and monitor believes there are 2.

By the way, the pilot threshold will probably need to be increased after the change I made to the pilot filter which allows more of the pilot signal through.

I wonder if perhaps the pilot detector might work better if the pilot samples were summed as complex numbers before computing the magnitude. Even if the pilot is off by 2 Hz, there shouldn't be much rotation in 20 ms.

ka9q, 2025-02-15:
I was also thinking that the pilot level has increased because of the widening of the filter.

ka9q, 2025-02-15:
Yeah, you're probably right. Opus probably works correctly because I keep the same payload type for mono and stereo, and monitor doesn't look at the channel count.

ka9q, 2025-02-15:
OK, I just pushed over some patches; give them a try. With WFM, selecting mono/stereo with control actually clears or sets a stereo enable flag, which defaults on. Because this is currently overloaded on the channel count command, clicking 'stereo' on a mono signal won't appear to do anything, but it will let the receiver switch to stereo if a pilot appears. I'll probably clean this up.

The actual channel count becomes 2 when stereo is enabled and a pilot is present, otherwise it's 1. In the PCM modes, separate payload types are dynamically allocated for stereo and mono, so there can be a glitch during the first switchover after 'monitor' starts because it may have to wait up to 500 ms for the status update to explain what the new payload type means. 'monitor' caches previously used payload types so further switching should not cause any glitching. Yes, it would have been easy to just send two channels for a mono station, but that's just too wasteful.

As before, Opus works just fine and I recommend it as the default for anything you are listening to by ear.

argilo, 2025-02-15:
It looks like this is working properly now, and monitor happily toggles between 1 and 2 channels as needed.

closed 2025-02-15 by argilo