ka9q/ka9q-radio #125: Not possible to keep WFM squelch open

argilo, 2025-02-14:
Setting something like `squelch-open = -100` doesn't keep the WFM squelch open. It seems the problem is that `chan->sig.snr` reaches zero (when the s/n reported by "monitor" is around 5 dB).

ka9q, 2025-02-15:
I don't understand the problem. I'm having no problem forcing the squelch open on wfm right now. But I did spot one bug that might be relevant. In the NBFM demodulator (fm.c), and previously in wfm.c, there are two squelches. The first divides the total inband power by the bandwidth times the estimated noise spectral density; that's the same SNR that appears in the Signal column. If that's below the squelch close threshold, the squelch stays closed.

Otherwise I compute the SNR with a FM-specific algorithm that relies on FM being constant envelope; any variance must be noise. I divide the average amplitude by its standard varianceides the average signal amplitude by its standard deviation. (FM is constant envelope, so any variance must be noise). Because this has a Rayleigh/chi-squared distribution, not Gaussian, there's a bias at low SNR that I correct with the function misc.c:fm_snr().

This works well for NBFM, but I recently took it out of the wfm demodulator because it's a little CPU intensive and squelches aren't heavily exercised in FM broadcasting anyway. But I somehow made the mistake of running the power SNR through fm_snr(). I'll take it out, but I don't understand why you're having trouble.

argilo, 2025-02-15:
Basically I see this whenever I tune a weak FM station, or when I tune to a frequency which has no station at all. The squelch will stay open for a while and the s/n ratio reported by monitor ticks down, and when it reaches about 5 the squelch kicks in. I added some printf debug statements and observed that the squelch closes when `chan->sig.snr` reaches zero.

ka9q, 2025-02-15:
I took out the spurious fm_snr() call; does this help? Yes, it can take time for the SNR to settle down after retuning because of the heavily damped N0 estimate. I suspect that HD Radio sidebands, if present, may also complicate the estimate.

I could always restore the moment-based squelch (average amplitude and its variance) but I'd rather not spend the CPU unless it's really necessary.

ka9q, 2025-02-15:
Yes, all those SNR figures should be the same since they come from the same source. The only remaining problem might be the long settling time needed for large changes in estimated N0. If that happens just from tuning within the FM broadcast band, it would be an artifact from noise-like HD-Radio sidebands. You say you don't have any near you but HD-Radio is common in southern California so I'll watch for it.