ka9q/ka9q-radio #126: Assign specifiq SDRplay device

Itsisme, 2025-02-14:
With the KA9Q and a RSP1A how can I assign a specifiq SDRPLAY device to KA9Q

As also other sdrplay devices are connected for other purposes I would like to know if thats possible

This to run auto_rx with KA9Q besides other software like OWRX+  

Chrs....

ka9q, 2025-02-15:
I don't actually know as K4VZ wrote sdrplay.c. I've tweaked it a little but I have very little documentation on the sdrplay.

Itsisme, 2025-02-15:
Thnxs, Would be a nice feature maybe in the future :)

cmh716, 2025-02-15:
Have you specified a serial number in your config on both softwares? I would expect this to work. I use ka9q with two different fobos devices on two different bands, and specify serial numbers to ensure the correct device is used. I think SDRPlay should work the same way.