ka9q/ka9q-radio #127: monitor + opus on Mac

cmh716, 2025-02-18:
If MacOS support is not desired—fair enough—we can just close this issue. However, I noticed that when using a monitor on my Mac, the CPU and performance are terrible when the audio source is opus.

For example, I set it up to listen to all the CB frequencies in am mode (I cannot seem to get amsq working reliably on CB) and using opus mode.  On Linux, monitor cb.local works well.  The menus are responsive, etc.  When playing on MacOS, the process uses a lot of CPU, but more importantly, the arrow keys are super laggy, so I cannot easily make changes.  When I switch to s16be, the CPU drops, and the menu is nice and responsive.  I assume this is due to a poor support for opus on MacOS and nothing to do with ka9q's program?

Does anyone else have this issue?  These are the version of libogg and opus I have installed:

libogg: stable 1.3.5 (bottled), HEAD
opus: stable 1.5.2 (bottled), HEAD

ka9q, 2025-02-18:
Hmm. I run monitor on my macbook all the time. Opus is my default format for channels I'm listening to by ear (e.g., FM repeaters, SSB/CW, but not channels for digital decoders). I don't seem to have this problem. I'm also using Opus 1.5.2.

cmh716, 2025-02-18:
Thanks Phil. You’re able to use the up and down arrows without lagging on the Mac when using opus? If so, it must be my setup then. s16be Worked perfectly for me.

ka9q, 2025-02-18:
Right, I do that all the time, sometimes with dozens of channels. I monitor all 2m/125cm/70cm repeaters audible in San Diego. I'm curious about what you find.

cmh716, 2025-02-18:
I think this is the cause, Phil:

https://github.com/ka9q/ka9q-radio/compare/81312bc6e87660918f78f2ab9044e723af9f32a4…69e7a51e8186c914ad116041990ad6c5d6b40306

I compiled at 81312bc6e87660918f78f2ab9044e723af9f32a4 and have the issue, checked out the previous 69e7a51e8186c914ad116041990ad6c5d6b40306 and no longer have the issue

I can easily reproduce the issue toggling back and forth

cmh716, 2025-02-18:
This is the test group that demonstrates the issue:

[Citizens Band]
disable = no
data = cb-am.local
encoding = opus
mode = am
freq = "26m965 26m975 26m985 27m005 27m015 27m025 27m035 27m055 27m065 27m075 27m085 27m105 27m115 27m125 27m135 27m155 27m165 27m175"
freq0 = "27m185 27m205 27m215 27m225 27m255 27m235 27m245 27m265 27m275 27m285 27m295 27m305 27m315 27m325 27m335 27m345 27m355"
freq1 = "27m365 27m275 27m385 27m395 27m405"

cmh716, 2025-02-18:
I notice on the newer version, the rates are slowly updating and show " 1110.0000"
On the older version, they update super fast and are tin the range of 8-10.0.  Maybe the problems exists on linux too, maybe it's just slower on MacOS.  The audio sounds flawless -- it's just the display.

cmh716, 2025-02-19:
It's not happening on Linux. It seems like some sort of MacOS ncurses issue where it's not reading memory correctly or something.  Every rate is showing as 1110.0000

ka9q, 2025-02-19:
[image]

Man, every CB channel at once, without squelch? You are a glutton for punishment. (To be honest, I've done it too.)
I just tried it, and it works fine on my macbook.

cmh716, 2025-02-19:
Yeah, it's quite the cacophony of roger beeps and nonsense, but amusing at times.
I have 2 screenshots.  One before the changes to monitor-display and the other after it

[image]
[image]

cmh716, 2025-02-19:
Looks like the rate becomes 0.000

cmh716, 2025-02-19:
This is what I did to test it:

cd /tmp
git clone https://github.com/ka9q/ka9q-radio.git 
cd ka9q-radio
git fetch --all
git checkout 69e7a51e8186c914ad116041990ad6c5d6b40306  
C_INCLUDE_PATH="$(brew --prefix)/include:$(brew --prefix)/opt/ncurses/include" LIBRARY_PATH="$(brew --prefix)/lib:$(brew --prefix)/opt/ncurses/lib" make
./monitor thechannel.local
(I have the 0.000 issue)

git checkout 81312bc6e87660918f78f2ab9044e723af9f32a4
make clean
C_INCLUDE_PATH="$(brew --prefix)/include:$(brew --prefix)/opt/ncurses/include" LIBRARY_PATH="$(brew --prefix)/lib:$(brew --prefix)/opt/ncurses/lib" make
./monitor thechannel.local
(it works fine)

cmh716, 2025-02-19:
brew info ncurses                                                                                                                         ─╯
==> ncurses: stable 6.5 (bottled) [keg-only]

cmh716, 2025-02-19:
I changed

      mvprintwt(y,x,"%*.*f",width,.001 * sp->datarate,sp->datarate < 1e6 ? 1 : 0); // decimal only if < 1000
to
      mvprintw(y, x, "%*.*f", width, sp->datarate < 1e6 ? 1 : 0, .001 * sp->datarate);

And it's working fine, but you're more proficient than I am with C, so I will defer to your judgement.

ka9q, 2025-02-19:
I do believe you're right! The mystery is how the original code works here, otherwise I would have fixed it long ago. I'm using 'port' rather than 'brew' but I think the underlying library should be the same.

ka9q, 2025-02-19:
Fix pushed. Thanks for finding this!

closed 2025-02-19 by ka9q

cmh716, 2025-02-19:
Awesome.  Thanks Phil!