ka9q/ka9q-radio #128: Discontinuous I/Q output when channel sample rate is an odd multiple of 200

argilo, 2025-02-20:
When the sample rate for an I/Q channel is set to an odd multiple of 200 Hz, discontinuities occur in the output every 20 ms. For instance:

```
./tune -r fm.local -s 1234 -R 500200 -f 87m5 -L -200000 -H 200000 -m iq
timeout 10 ./pcmrecord -c -r -S 1234 fm-pcm.local > 500200.cs16
inspectrum 500200.cs16
```

[image]

I suspect there is a phase discontinuity between each 20 ms block.

There are no discontinuities when an even multiple of 200 Hz is used instead.

This issue is not urgent. I just wanted to log it so that it's not forgotten.

ka9q, 2025-02-21:
You sure you're using all my latest patches on this? The problem was originally quite audible when I used oddball sampling rates like 48,200 Hz, which I never had reason to test before. Nor could I have, until I used libsamplerate in 'monitor'.

The problem occurred when odd IFFT lengths were used. The audible glitches went away when I rounded odd IFFT sizes up by 1 to even lengths, keeping the same number L of output samples per block (and thus the requested sample rate) while making the impulse response duration M an odd number.

In fast convolution the I/FFT length N  = L + M - 1, where M-1 is the number of overlapped samples per block and the number of samples in the desired filter impulse response is M. These numbers are scaled by the decimation ratio between the input and output sample rate, which is why the output sample rate must be a multiple of the block rate (so an integral number of samples are generated every block time) *and* a multiple of the FFT bin spacing (so an integral number of frequency bins and time domain samples are generated during each block time.) For a 20 ms block time (50 Hz block rate) and 40 Hz bin spacing (4/5 * 50 = 40), any multiple of 200 Hz should be OK.

argilo, 2025-02-21:
> You sure you're using all my latest patches on this?

Yes, the screenshot I posted above was taken while I was running the most recent commit.

I'll try a few more tests to see whether I can observe any change in behaviour before & after the relevant code changes.

argilo, 2025-02-21:
It does appear that the code changes made an improvement. When I tested with a single tone as input, I no longer saw the 72-degree phase jumps after the changes. But I still see (smaller) blips in the spectrogram, so there seems to be a discontinuity of some kind still remaining.

argilo, 2025-02-21:
Update: With the tone near the recording's center frequency, there are no phase jumps. But if the tone is far from the center frequency, large phase jumps are visible.

argilo, 2025-02-21:
[image]

argilo, 2025-02-21:
I see similar behaviour listening to an upper-sideband demadulator:

Prior to the code changes, I hear a loud 50 Hz buzz regardless of the frequency of the tone I'm listening to.

After the code changes, I don't hear much buzz when the tone is at a low frequency, and I hear more buzz as the tone frequency increases.

ka9q, 2025-02-21:
Now that's an interesting clue.

argilo, 2025-02-21:
I guess it would suggest that there's now a time jump between blocks, rather than a phase jump.

ka9q, 2025-02-21:
I think I understand what's going on. I can't round up the FFT block size because that slightly decreases the width of each FFT bin slightly below 40 Hz. You don't notice this near the center, but the shift accumulates at the edges. I'll have to figure out how to properly handle odd blocksizes, probably by correctly handling even-length FIR filters.

ka9q, 2025-02-21:
I think that's it. I removed the round-up-to-even, selected the Liquid DSP library's Remez FIR design algorithm (code already in filter.c but disabled) and I can't hear any obvious artifacts at, e.g, 48200 Hz. If you want to try it before I push it over, you'll have to add -lliquid in the appropriate places in the makefile.

I must have a bug in my Kaiser-windowed FIR design routine in handling even filter lengths. I will still try to fix it.

ka9q, 2025-02-21:
Never mind, it's still broken even with liquid filter generation. I forgot I had to listen in a phase sensitive mode (IQ); the phase jumps don't bother envelope detection.

ka9q, 2025-02-21:
Can you tell me exactly what it was doing to each block with odd multiples of 200 Hz and the original odd-length IFFT? Was it a periodic timing jump? I'm still trying to understand the math here.

argilo, 2025-02-21:
I believe there was a phase jump (of ~72 degrees ?) between each block.

argilo, 2025-02-21:
I could imagine there might be a misalignment of one sample or a half a sample somewhere. For instance, I noticed that when `M` is even, the Kaiser window generated by `window_filter` has its center halfway between two taps, whereas the filter in `buffer` has its center exactly at index 0. So when they're combined in `buffer[n] = buffer[(n-M/2+N)%N] * kaiser_window[n] * gain;` the two are off by 1/2. I don't think that's the cause of the buzzing, though.

ka9q, 2025-02-22:
I looked at my Kaiser window code, thinking much the same thing. But when I used the liquiddsp library to design my filter, the results were the same.

argilo, 2025-02-22:
Indeed.

The part that attracted my attention was the `M/2`. I suppose any other code that divides `M` or `N` by two could be a potential source of bugs that would only affect odd multiples.

argilo, 2025-02-22:
Another observation: With the original code, I recorded a tone at 24000 and 24200 sps. The one recorded at 24200 appears to be about 1% lower in frequency.

If I repeat the experiment with 24000 & 24400 sps, the measured frequency is the same.

argilo, 2025-02-22:
The frequency measurements differ in this way even if I confine the measurement to a single 20-ms block, avoiding the discontinuities.

It would seem that somehow only 480 samples from each 484-sample / 20-ms block are being used, and 4 samples skipped.

ka9q, 2025-02-22:
Very interesting. I assume the sample counts are correct?

ka9q, 2025-02-22:
Just to confirm, this is with the code that doesn't round the IFFT size up to an even number? My current commits disallow (temporarily, I hope) sample rates that are an odd multiple of 200 Hz, so you'll have to comment out the relevant checks for your tests.

argilo, 2025-02-22:
> I assume the sample counts are correct?

I'm not certain about the 480. But it seems like a plausible number to explain the frequency drop.

> Just to confirm, this is with the code that doesn't round the IFFT size up to an even number?

Yes, this was tested with commit ad7a759028acb1e90c01e04a9be8b28b2042a92c, which is prior to the rounding change.

argilo, 2025-02-22:
I did some more testing with multiple tones, and it seems like the frequency isn't dropping by a fixed percentage, it drops by ~50 Hz. I suppose that could be caused by the fine frequency offset being wrong.

argilo, 2025-02-22:
If I replace `shift` with `(shift-1)` on this line, then the phase discontinuities disappear for odd multiples of 200 Hz (and appear for even multiples):

https://github.com/ka9q/ka9q-radio/blob/4025a34db6e88dce87b8f67c7eb9cc339b920261/radio.c#L612

But the frequency is still wrong.

argilo, 2025-02-22:
I see that `execute_filter_output` computes `slave->bins/2` in many places, which could mean that it implicitly assumes the number of bins is even. If I replace that with `(slave->bins+1)/2` so that it rounds in the opposite direction when the number of bins is odd, then the phase discontinuities remain but the received frequency is shifted up instead of down.

argilo, 2025-02-22:
So, my best guess at this point is that the phase & frequency adjustments need to take into account that the "center" of the slave FFT is going to be off by half a bin when the number of bins is odd.

ka9q, 2025-02-22:
OK, I think I am beginning to see what is happening. Right, now I refer to the most negative frequency bin as N/2 and the most positive as N/2 - 1. (This is in non-rotated spectra, which I use). This is correct for N even. It classifies the Nyquist bin as the most negative frequency, but I'm not sure it matters because I'm pretty sure all my filter responses are zero there, or they certainly should be. . , whchafter reading some stuff about FFTs with even and odd Ns, I think I should refer to the most negative frequency bin as (N+1)/2 and the most positive as (N-1)/2. For N odd this will give an equal number of positive and negative frequency bins, plus the DC bin. For N even, this will make the Nyquist bin (which you shouldn't use) the most negative frequency, which is what I think I already do. The problem

ka9q, 2025-02-22:
I think I got it. I rewrote much of filter.c:execute_filter_output(), paying careful attention to bin indexing with both even and odd sizes, and it now seems to work fine with odd multiples of the sample rate (which typically means an odd IFFT size). I'd been meaning to rewrite this anyway to merge the multiplication of the frequency response into the copy loop to avoid another pass over the data. I'm starting to get very aware of memory access patterns and cache hit ratios.

But I won't declare victory until you test it. I've tested it with an AirspyHF, which uses a complex->complex filter, and an rx888, which uses a real->complex filter with an upright spectrum. I guess I should also test an Airspy R2, which uses real->complex with an inverted spectrum.

argilo, 2025-02-22:
Sounds good. I'll take it for a spin tonight.

ka9q, 2025-02-22:
And it's working on the Airspy R2, which I use to monitor 2m/125cm/70cm FM. I was momentarily confused when I switched to a 24200 Hz sampling rate and everything seemed to freeze for a few seconds. Then I remembered that I create a moderate-effort FFTW plan for each new block size, and this takes a few seconds on a Pi4. The log told the story.

argilo, 2025-02-23:
Looking good with an Airspy R2! I tried piping samples into nrsc5 with various odd and even multiples of 200 SPS, and I always get ~37 dB MER, 0 BER.

argilo, 2025-02-23:
I also tried removing libsamplerate from the nrsc5 wrapper and feeding in samples at 744200 Hz. There's a small drop in MER (to ~33.5 dB), presumably due to the residual sample rate error, and nrsc5's CPU utilization drops significantly (from ~11% to ~2.7%). radiod doesn't seem to be using much more CPU, so it might make sense to leave the burden of an oddball FFT size on it, instead of using a resampler.

argilo, 2025-02-23:
One thing I noticed while listening to a tone with a USB demodulator: as I increase the sample rate (10000, 20000, 40000, 80000) the frequency of the tone increases somewhat. Maybe there's an error somewhere in the fine frequency adjustment?

argilo, 2025-02-23:
I also had a look over spectrograms of signals recorded at even and odd sample rates, and don't see any evidence of discontinuities.

ka9q, 2025-02-23:
How are you listening? 'monitor' uses libsamplerate, and I've noticed a transient change in the output pitch when I change the input sample rate. I don't see how or why my demodulated frequencies should change.

argilo, 2025-02-23:
I was listening with monitor.

ka9q, 2025-02-23:
I'm listening to WWV on 15 MHz with two USB channels on the same receiver. Both are emitting s16be output (since opus only supports a few standard rates). When I change the sample rate of a channel, I hear a very brief pitch transient but the two channels quickly resynchronize.

ka9q, 2025-02-23:
I think the transient is caused by the timing skew between the RTP data and the status packet that explains what the RTP payload type means. So it's expected. The delays also change but that's also expected given all the buffering.

argilo, 2025-02-23:
I repeated the experiment with pcmrecord, measuring the frequency with Audacity.

10000 SPS: 709 Hz
20000 SPS: 718 Hz
30000 SPS: 726 Hz
40000 SPS: 736 Hz

argilo, 2025-02-23:
Same result for a higher-frequency tone. ~9 Hz tone increase for each 10000 Hz sample rate increase.

ka9q, 2025-02-23:
Hmm, I haven't heard this yet but I'll test some more. The error is so small (<< 1 bin) that I suspect some sort of floating point precision problem. The complex oscillator code uses double precision for this reason. But then why is it happening?

ka9q, 2025-02-23:
OK, I do notice a change in the output tone frequency, but jiggling the radio tuning (e.g., up and down 1 Hz) brings it back in sync. I suspect I need to reinit something when I change the sample rate. Not immediately obvious what it is since I restart the receiver channel whenever the output sample rate changes. And then, I still hear a much slower beat between the two audio tones, less than 1 Hz. Maybe some numerical roundoff somewhere, but not necessarily in radiod; it might be in libsamplerate. I don't hear any residual beat at all when I use standard sample rates, e.g., 12 kHz and 48 kHz.

ka9q, 2025-02-23:
Found it. I needed to reinitialize chan->filter.remainder = NAN when the demodulator restarts after the sample rate change, otherwise it wasn't getting reinitialized properly until you moved the tuning. There's even a comment about it at line 601 in radio.c.

argilo, 2025-02-23:
Everything looks good now, so I'll close off the issue.

closed 2025-02-23 by argilo

ka9q, 2025-02-23:
Thanks for all your help on this! Greatly appreciated.