ka9q/ka9q-radio #129: When building on Ubuntu 22.04 complaining about SDRPLAY_RSPdxR2_ID (20/2/2025)

waterwin, 2025-02-20:
Been building on various systems the last couple of days and this is the first time this happened.

'... did you mean SDRPLAY_RSPdx_ID'

cmh716, 2025-02-20:
I just tried on 24.04 and did not get any warnings.  Does it have more detail?

No errors on:
cc -c -fpic -DNDEBUG=1 -O3 -march=native -std=gnu11 -pthread -Wall -funsafe-math-optimizations -fno-math-errno -fcx-limited-range -D_GNU_SOURCE=1 -Wextra -MMD -MP  -o sdrplay-pic.o sdrplay.c
cc -shared -o sdrplay.so sdrplay-pic.o -lsdrplay_api

I am on the latest ka9q and latest SDRPlay driver

waterwin, 2025-02-20:
cc -c -fpic -DNDEBUG=1 -O3 -march=native -std=gnu11 -pthread -Wall -funsafe-math-optimizations -fno-math-errno -fcx-limited-range -D_GNU_SOURCE=1 -Wextra -MMD -MP  -o sdrplay-pic.o sdrplay.csdrplay.c: In function ‘set_antenna’:sdrplay.c:762:34: error: ‘SDRPLAY_RSPdxR2_ID’ undeclared (first use in this function); did you mean ‘SDRPLAY_RSPdx_ID’?  762 |   } else if(sdr->device.hwVer == SDRPLAY_RSPdxR2_ID){      |                                  ^~~~~~~~~~~~~~~~~~      |                                  SDRPLAY_RSPdx_IDsdrplay.c:762:34: note: each undeclared identifier is reported only once for each function it appears insdrplay.c: In function ‘get_lna_states’:sdrplay.c:882:34: error: ‘SDRPLAY_RSPdxR2_ID’ undeclared (first use in this function); did you mean ‘SDRPLAY_RSPdx_ID’?  882 |   } else if(sdr->device.hwVer == SDRPLAY_RSPdxR2_ID){      |                                  ^~~~~~~~~~~~~~~~~~      |                                  SDRPLAY_RSPdx_IDsdrplay.c: In function ‘set_notch_filters’:sdrplay.c:1084:34: error: ‘SDRPLAY_RSPdxR2_ID’ undeclared (first use in this function); did you mean ‘SDRPLAY_RSPdx_ID’? 1084 |   } else if(sdr->device.hwVer == SDRPLAY_RSPdxR2_ID){      |                                  ^~~~~~~~~~~~~~~~~~      |                                  SDRPLAY_RSPdx_IDsdrplay.c: In function ‘set_biasT’:sdrplay.c:1102:34: error: ‘SDRPLAY_RSPdxR2_ID’ undeclared (first use in this function); did you mean ‘SDRPLAY_RSPdx_ID’? 1102 |   } else if(sdr->device.hwVer == SDRPLAY_RSPdxR2_ID){      |                                  ^~~~~~~~~~~~~~~~~~      |                                  SDRPLAY_RSPdx_IDsdrplay.c: In function ‘show_device_params’:sdrplay.c:1276:34: error: ‘SDRPLAY_RSPdxR2_ID’ undeclared (first use in this function); did you mean ‘SDRPLAY_RSPdx_ID’? 1276 |   } else if(sdr->device.hwVer == SDRPLAY_RSPdxR2_ID){      |                                  ^~~~~~~~~~~~~~~~~~      |                                  SDRPLAY_RSPdx_IDmake: *** [Makefile:249: sdrplay-pic.o] Error ***@***.***:~/ka9q-radio$

 
  

   

|  | Virus-free.www.avast.com |

cmh716, 2025-02-20:
I am on this version of the SDRPlay API: SDRplay_RSP_API-Linux-3.15.2.run
Which version are you on?

waterwin, 2025-02-20:
Chris I have no idea. I do not use my SDRplay with radiod and it has never been near this pc. I do not manually install anything from SDRplay either. In /usr/local/lib I find libsdrplay_api.so and so.3 and so.3.14

cmh716, 2025-02-20:
Oh, so that is probably why. Phil recently added changes that if it see the SDRPlay or Fobos drivers installed, it will compile support for them.  At one point, you had them installed. You could remove the drivers and it will not compile support for SDRPlay.

cmh716, 2025-02-20:
You'll need to fully remove /usr/local/include/sdrplay*

waterwin, 2025-02-20:
Might have been from another install like OpenWebRx. I used Uninstall SDRplay API 3.06.1 from an Raspberry Pi 4 running Raspbian buster ($1926800) · Snippets · GitLab

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Uninstall SDRplay API 3.06.1 from an Raspberry Pi 4 running Raspbian bus...

GitLab.com
 |

 |

 |

changing .06 to 14
make now has no more issues and finalizes the fobos actions
Thanks again for fast and effective support
Erwin

ka9q, 2025-02-21:
I've encountered this myself when the header files were incomplete or out of date. Maybe I should look for another, newer SDRPlay header file since there's more than one.

ka9q, 2025-02-23:
I don't use the SDRPlay that much, and I know little about the different versions of its driver package for Linux. Can someone suggest the best component file I should look for to see if it's installed? Ideally it would be one added relatively recently so I won't try to build with an old unusable version.

ka9q, 2025-02-24:
I'm closing this unless I get some suggestions on what I should change in the Makefile to avoid compilation errors on a partial or outdated SDRPlay driver installation.

waterwin, 2025-02-24:
Maybe just a note in the Docs?

To be aware of the effect of old drivers.
Erwin