ka9q/ka9q-radio #132: Heap user-after-free when monitor queue grows too large

argilo, 2025-02-23:
ASAN detects a heap use-after-free in monitor when the queue fills up and is emptied. The problematic code is:

https://github.com/ka9q/ka9q-radio/blob/59d61a2610482b8e736dd577c2a797c7d9d7678b/monitor-data.c#L157-L170

Lines 157-163 discard all the packets from the queue, but `q_prev` still points to one of them. Then on line 170, `qprev->next` is accessed, triggering the use-after-free.

Perhaps an appropriate fix would be to add `q_prev = NULL` to the code block that empties the queue.

```
=================================================================
==1269533==ERROR: AddressSanitizer: heap-use-after-free on address 0x531007274800 at pc 0x60f83e08e01b bp 0x784e302f5690 sp 0x784e302f5680
WRITE of size 8 at 0x531007274800 thread T9
    #0 0x60f83e08e01a in dataproc /home/argilo/git/ka9q-radio/monitor-data.c:170
    #1 0x784e4445ea41 in asan_thread_start ../../../../src/libsanitizer/asan/asan_interceptors.cpp:234
    #2 0x784e4369caa3 in start_thread nptl/pthread_create.c:447
    #3 0x784e43729c3b in clone3 ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

0x531007274800 is located 0 bytes inside of 65648-byte region [0x531007274800,0x531007284870)
freed by thread T9 here:
    #0 0x784e444fc4d8 in free ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:52
    #1 0x60f83e08de2b in dataproc /home/argilo/git/ka9q-radio/monitor-data.c:162
    #2 0x784e4445ea41 in asan_thread_start ../../../../src/libsanitizer/asan/asan_interceptors.cpp:234
    #3 0x784e4369caa3 in start_thread nptl/pthread_create.c:447

previously allocated by thread T9 here:
    #0 0x784e444fd9c7 in malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:69
    #1 0x60f83e08c09b in dataproc /home/argilo/git/ka9q-radio/monitor-data.c:79
    #2 0x784e4445ea41 in asan_thread_start ../../../../src/libsanitizer/asan/asan_interceptors.cpp:234
    #3 0x784e4369caa3 in start_thread nptl/pthread_create.c:447

Thread T9 created by T0 here:
    #0 0x784e444f51f9 in pthread_create ../../../../src/libsanitizer/asan/asan_interceptors.cpp:245
    #1 0x60f83e087ad3 in main /home/argilo/git/ka9q-radio/monitor.c:395
    #2 0x784e4362a1c9 in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #3 0x784e4362a28a in __libc_start_main_impl ../csu/libc-start.c:360
    #4 0x60f83e084cc4 in _start (/home/argilo/git/ka9q-radio/monitor+0x67cc4) (BuildId: 37f5f2c209a875feaee68fd553e2d3a9f15fab32)
```

ka9q, 2025-02-23:
Ah, this one has been bugging me for a long time! I'll try your fix.

closed 2025-02-23 by ka9q

argilo, 2025-02-23:
Looks like it works.

By the way, ASAN & UBSAN output can be redirected to a file, which is particularly useful for ncurses applications. Just prefix the command with `UBSAN_OPTIONS="log_path=ubsan.log"`. (There is a `ASAN_OPTIONS` environment variable as well, but it seems that `UBSAN_OPTIONS` is used when ASAN and UBSAN are used together.)