ka9q/ka9q-radio #135: ft8.log : execlp(decode_ft8) returned errno 2 (No such file or directory)

waterwin, 2025-02-25:
radio with Fobos fresh installed
ft8-decoded.service running
/etc/radio populated
/var/lib/ka9q-radio/ft8 available
/usr/local/bin/jt-decoded available

/var/log/ft8.log is only receiving this error line, no decodes like before

Using commit from 2 weeks ago (18 feb) as since then some new functionality has been introduced and the software is still under construction for some of these

No clue

ka9q, 2025-02-26:
decode_ft8 is actually provided by a separate package, ka9q/ft8_lib. It's a slightly modified fork of kgoba/ft8_lib.

I plan to completely rewrite the ka9q-radio ft8/ft4 decoding stuff to use the 'jt9' command from the wsjtx package soon, so I don't really want to spend a lot of time on this.

waterwin, 2025-02-26:
I understand that and thank you for your reply. It is only that I have various ka9q-radio setups working well and when installing fresh I run into strange new things. Should be good for my education.

closed 2025-02-26 by waterwin