ka9q/ka9q-radio #137: Fobos clk_source external - confirmation - tested - working

waterwin, 2025-02-28:
Today I tested the Fobos SDR with external 10 MHz clock source (RFZero).

Works! 

F4VTQ/K is a KiwiSDR with GPS controlled clock
F4VTQ/S is Fobos SDR with RFZero GPSDO with its GPS controlled clock
F4VTQ/R is Red Pitaya 122/16 with GPS controlled clock

y-m-d local | txCall | txGrid | rxCall | rxGrid | MHz | W | SNR | drift | k | txAz° | mode | k/W | spotQ | version
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2025-02-28 12:24 | YO6FFW | KN25sp | F4VTQ/K | JN28ux | 21.096095 | 0.1 | -18 | 0 | 1536 | 291 | WSPR2 | 15360 | 7461 | WD_3.2.3-1
2025-02-28 12:24 | YO6FFW | KN25sp | F4VTQ/S | JN28ux | 21.096095 | 0.1 | -20 | 0 | 1536 | 291 | WSPR2 | 15360 | 6583 | WD_3.3.1-1
2025-02-28 12:24 | YO6FFW | KN25sp | F4VTQ/R | JN28ux | 21.096094 | 0.1 | -16 | 0 | 1536 | 291 | WSPR2 | 15360 | 8338 | RP122-16

cmh716, 2025-02-28:
That's great.  I have a Web-888 SDR and considered trying the clock output with the Fobos SDR

ka9q, 2025-02-28:
Excellent!

ka9q, 2025-03-02:
I'm assuming nothing has to be fixed here.

closed 2025-03-02 by ka9q