ka9q/ka9q-radio #139: I'm not getting data with samprate=768k

AndyLinwood, 2025-03-07:
I'm trying to run WebSDR the same way as in Maasbree Netherlands using the rx 888. My Radiod instance is working fine with samprate =768k in the .conf file. But it doesn't send the data to the multicast. And because websdr is configured to receive the raw iq stream via fifo, it does not start.
In the file Radiod@rx888.conf looks something like this

```
[WEBSDR_768] 
disable = no 
data = websdr768-pcm.local 
mode = iq 
freq = "3700k 5300k 7125k 14150k 21200k 28350k 28850k 29350k" 
samprate = 768k 
low = -383k
high = 383k 
agc = 0 - We do now want to enable AGC if we wish to have a calibrated S-meter.
gain = 20 - Th
```
What could be causing this?
I hope you've explained it clearly?

ka9q, 2025-03-08:
Don't use pcmcat, it's obsolete. Use 'pcmrecord [-v] --stdout --ssrc <ssrc> <mcast-group>'. The old pcmcat assumed one of a small set of predefined sample rates and formats statically encoded on 7-bit RTP payload types, so nonstandard ones didn't work. The new pcmrecord looks at the status messages to learn the sample rate and format associated with a given dynamically allocated payload type.

Make sure the output encoding isn't 'opus'; radiod will refuse to send anything unless the sample rate is one of the standard supported rates (8k, 12k, 16k, 24k, 48k). But I suspect that's not your problem as the default is 's16be'.

Take a look at the --exec option to pcmrecord. It executes a specified command for each new ssrc that appears on a multicast group, with arguments for sample rate, frequency, etc. It can often significantly simplify configuration.

closed 2025-03-08 by ka9q

AndyLinwood, 2025-03-10:
Thank you. I understand you