ka9q/ka9q-radio #140: command duplication in monitor

jonsnow1357, 2025-03-08:
Hi,

It seems that when using monitor C (SHIFT + c) and (SHIFT + right arrow) are doing the same operation.

Best regards

ka9q, 2025-03-08:
Not sure what you mean. 'C' toggles between constant and adaptive playout times. shift + right-arrow increases the playout buffer 1 millisecond. I tested it and looked at the source (especially for accidental switch fall-throughs) and didn't see anything out of the ordinary, though I did fix two comments. I also checked the help file.

Can you describe more fully what you're seeing?

jonsnow1357, 2025-03-09:
hmmm  ... could my my setup then

If I press 'C" I see "Constant delay" showing up and disappearing in the CLI
If I press 'SHIFT + arrow right' I see the same thing .

I am working over SSH on a remote laptop, let me read a bit about TERM and env variables ...

ka9q, 2025-03-12:
OK, maybe your laptop isn't generating what it should when you hit an arrow key. Things weren't well standardized before Unicode, and I'm not sure everybody follows it even now.

closed 2025-03-12 by ka9q