ka9q/ka9q-radio #151: Build fails on macOS

argilo, 2025-05-02:
The addition of `-fcx-limited-range` in ddbcb11e2f106bad03499795d6086eb8325c48e5 broke macOS CI, because that option was not added to Clang until version 18. The GitHub actions runner has Clang 15.

What would you like to do about this?

jonsnow1357, 2025-05-09:
you can see this in the failed test for the documentation PR also :(

ka9q, 2025-05-09:
Seems like every change I make breaks the macos builds! The machine in front of me is almost always my Macbook, and I do make sure things compile locally even if I rarely run anything besides 'control' and 'monitor' locally; everything else gets pushed to Linux machines that actually have the radios. So obviously my environment is different from whatever CI is using, and I have no idea how to change it because, quite frankly, I don't see it as a particularly useful feature. It just fills my mailbox with failure messages.

argilo, 2025-05-09:
It's probably useful to know that ka9q-radio fails to build on Clang versions earlier than 18, which I expect is the default compiler on most macs. Are you perhaps using gcc instead?

ka9q, 2025-05-09:
This is odd:
bash-5.2$ cc --version
Apple clang version 17.0.0 (clang-1700.0.13.3)
Target: x86_64-apple-darwin24.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

yet it all works for me.

closed 2025-06-24 by ka9q