ka9q/ka9q-radio #153: ALE Decoder Request

commsman98, 2025-05-05:
Would it be possible to include an ALE decoder as it would be another useful mode for monitoring HF Propagation.

It looks like work had started on an ALE decoder a number of years ago which was called LinuxALE https://github.com/DigitalHERMES/LinuxALE

It would be a great addition to the excellent ka9q-radio

Thank you

ka9q, 2025-05-08:
I have never used ALE, but if you'd like to integrate that one with ka9q-radio, be my guest! I provide the **pcmrecord** utility as a general-purpose link to application programs; not only can it record to a file (as the name suggests) but it can also write to a pipe or, in its most general mode, invoke a user-specified command and pass the channel parameters to it, eg, sample rate, channel frequency and so on. I've just used it that way to feed the **dumphfdl** package for monitoring ARINC HFDL.

commsman98, 2025-05-08:
Thanks for the response Phil. Integrating the Linux ALE decoder with ka9q-radio is outside of my ability, but perhaps if we can leave this request on here someone else might take up the challenge!

It's unclear how well this LinuxALE works considering it was last updated 5 years ago, but I've been unable to find anything else for Linux. Most people seem to use PCALE which is Windows software. 

I've started to decode HFDL myself using ka9q-radio and it's working well. 

Thanks for all your work.

ka9q, 2025-05-09:
Excellent, glad to see someone else trying HFDL. I think you're the first after me. Note that I restructured it a few days ago, with a single systemd service (hfdl) and a helper shell script that passes the right channel list to dumphfdl depending on the channel center frequency passed by pcmrecord. It still seems much too messy so I might try to clean it up further. Right now, if they were to add a new channel I'd have to edit several different places. Ideally it should only be one, or even better, write a script to extract everything from the system configuration file that lists all the ground stations and their frequencies.

ka9q, 2025-05-09:
If you were to create a separate 12 kHz ka9q-radio channel with its own instance of dumphfdl, I'd be interested in a performance comparison with my present method of creating one ka9q-radio channel and dumphfdl instance for every band, with dumphfdl further demuxing the channels in each band. Both methods work, but I haven't really compared their performance.
Each has its advantages. The per-channel method is much cleaner (all the ka9q-radio channels have the standard 12 kHz sample rate) but requires 12 kHz of samples to carry one SSB channel because the dumphfdl decoder still requires an IQ input. (It seems like it should be easy enough to modify it to accept a single ouput at, say, 8 kHz, but it's his code.) And of course it fills up your process table with a lot of pcmrecord and dumphfdl processes.
The per-band method I'm using now is obviously much more ad-hoc, as I had to pick a sample rate for each band depending on its size and also to be efficient with ka9q-radio. But there are many fewer instances of pcmrecord and dumphfdl, and the I/Q sample rate per channel can be lower when there are a lot of closely spaced channels on a band.

commsman98, 2025-05-10:
Phil, it was an amateur radio friend who configured ka9q-radio with HFDL for me. It's helped me work out things like the need for a USB 3 card in my PC so I get the full 5 Gbps from a USB 3 port to run the RX888. I've only just started to decode HFDL but I'm not receiving as many stations or frequencies as I'd have expected. The biggest challenge is reducing noise levels across the HF band. 

My setup is: Wellgood Loop > RFComms Choke > Biast-t > Turn Island Filter > RX888 MK2. I use twisted pair cable from the Loop as I find lower noise levels than when I use coax. I've also added ferrites at various points but noise levels remain high. Here's a screenshot of what the HF band currently looks like [https://imgur.com/a/Y7FAqF1](url) 
The noise between 8.5MHz to 12MHz is VDSL and from around 19MHz is a characteristic of the Wellgood loop which other owners experience. 

I was hoping ALE decoding could be another mode to monitor multiple frequencies - the USAF are one example of a user with frequencies across the HF band [https://www.udxf.nl/[USAF-HFGCS.pdf](https://www.udxf.nl/USAF-HFGCS.pdf)](url)

My end goal is to use ka9q-radio with HF voice recording software called JSquelch [](https://github.com/jontio/JSquelch) to record multiple aero frequencies. Hopefully that will be achievable one day!

If / when I get the noise levels down I will experiment further with ka9q-radio and HFDL.

jonsnow1357, 2025-05-24:
@commsman98 you can use pcmrecord (already included) to record whatever you monitor with radiod. I hope to update some docs soon ...

ka9q, 2025-08-07:
Can you hear the HFDL signals by ear? Obviously if the signals are overwhelmed by noise you are not going to decode them with *any* receiver, so as the saying goes the antenna is the most important part of any radio station. (And I can say this as a computer, software and SDR guy, because it's still true.) Do you have any other way to decode HFDL, like a KiwiSDR? Comparisons are always helpful.

Sorry for the long delay in answering, this has been a busy time of year for us. The Dayton hamvention, the Friedrichshafen ham event, plus family visits and other related travel...

commsman98, 2025-09-06:
Sorry Phil but I'm no longer testing HFDL. 

Happy for this request to be closed as appreciate its not part of the ka9q-radio build.

Should anyone be able to help with linking JSquelch & ALE decoder to ka9q-radio then I can be contacted via my website [www.airband-radio.co.uk](url) 

Thank you

closed 2025-09-06 by commsman98