ka9q/ka9q-radio #154: measured sample rate error

HB9VQQ, 2025-05-05:
I am running ka9q-radio as part of the wsprdaemon software on a Proxmox hypervisor in a virtual machine. The RX 888 is connected to a USB3 port on the hypervisor and connected to the VM via USB passthrough. I wonder what is causing these error messages

May 05 20:50:54 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 57,995,928.3 Hz vs nominal 64,800,000 Hz
May 05 20:51:04 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,000,621.9 Hz vs nominal 64,800,000 Hz
May 05 20:51:14 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,005,121.0 Hz vs nominal 64,800,000 Hz
May 05 20:51:24 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,010,849.7 Hz vs nominal 64,800,000 Hz
May 05 20:51:34 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,011,986.3 Hz vs nominal 64,800,000 Hz
May 05 20:51:44 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,021,280.4 Hz vs nominal 64,800,000 Hz
May 05 20:51:54 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,023,469.6 Hz vs nominal 64,800,000 Hz
May 05 20:52:04 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,029,398.0 Hz vs nominal 64,800,000 Hz
May 05 20:52:14 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,037,203.2 Hz vs nominal 64,800,000 Hz
May 05 20:52:24 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,042,306.7 Hz vs nominal 64,800,000 Hz
May 05 20:52:34 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,046,809.2 Hz vs nominal 64,800,000 Hz
May 05 20:52:44 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,048,127.1 Hz vs nominal 64,800,000 Hz
May 05 20:52:54 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,051,464.5 Hz vs nominal 64,800,000 Hz
May 05 20:53:04 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,054,095.7 Hz vs nominal 64,800,000 Hz
May 05 20:53:14 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,057,233.7 Hz vs nominal 64,800,000 Hz
May 05 20:53:24 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,058,738.0 Hz vs nominal 64,800,000 Hz
May 05 20:53:34 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,064,346.9 Hz vs nominal 64,800,000 Hz
May 05 20:53:44 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,071,354.9 Hz vs nominal 64,800,000 Hz
May 05 20:53:54 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,069,737.6 Hz vs nominal 64,800,000 Hz
May 05 20:54:04 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,069,447.0 Hz vs nominal 64,800,000 Hz
May 05 20:54:14 ka9q-radio2 radiod@rx888-wsprdaemon[115276]: RX888 measured sample rate error: 58,072,284.5 Hz vs nominal 64,800,000 Hz

Thank you
Roland

ka9q, 2025-05-08:
The driver computes the total number of samples since startup and computes the average sample rate using the system clock. If it's been at least 1 minute since startup, and if the average rate differs from that expected by more than 1%, it prints that message. This is *almost always* due to data loss in the USB subsystem; that's why the indicated sample rate is less than nominal.
It's conceivable that buffering jitter or a real time clock adjustment by the Network Time Protocol daemon could trigger spurious messages, that's why I average over the entire time the rx888 has been running, and with a relatively liberal error threshold of 1%.
In your case I strongly suspect it's because you're running under a hypervisor. I never run real time programs in virtual machines because I don't trust their scheduling.

HB9VQQ, 2025-05-08:
Thank you. I have abandoned the idea of running radio on a hypervisor. Instead, I am using the USB4 port (40Gbps) and running 3 instances of radio on the same PC hardware https://groups.io/g/wsprdaemon/topic/multiple_instances_of_radiod/112665489

closed 2025-05-08 by HB9VQQ

rrobinett, 2025-05-09:
> Thank you. I have abandoned the idea of running radiod on a hypervisor.
> radiod on the same PC hardware

Hi Roland,

So do I understand correctly that you have 3 instances of KA9Q-radio
running on a bare metal Ubuntu 24.04 server?
Do you see any clock error messages from those radiod in the system log?
Also, are there any errors reported by wd-pcmrecord-errors?  It should
look like this:

***@***.***:~/wsprdaemon$ wd-pcmrecord-errors
0:  /dev/shm/wsprdaemon/recording.d/KA9Q_0
1:  /dev/shm/wsprdaemon/recording.d/KA9Q_0_WWV
There are multiple KA9Q receivers.  Which of their pcmrecord-errors.log
files do you want to view? [0-1] (default = 0) =>
Run tail -f /dev/shm/wsprdaemon/recording.d/KA9Q_0? [Yn] =>
=========== Starting at Thu May  8 22:11:30 UTC 2025 =================

=============== Thu 08 May 2025 22:19:12 UTC: ERROR: start ================
=========== Starting at Thu May  8 22:19:13 UTC 2025 =================
=========== Starting at Fri May  9 01:24:01 UTC 2025 =================

=============== Fri 09 May 2025 01:24:49 UTC: ERROR: start ================
=========== Starting at Fri May  9 01:24:51 UTC 2025 =================

HB9VQQ, 2025-05-09:
See my answer on groups.io