ka9q/ka9q-radio #163: Gain or AGC for Non Linear modes using RTL-SDR

pfassberg, 2025-07-01:
I'm trying to record a few FM broadcast channels using a RTL-SDR (RTL2832U).  The signal strength is unfortunately low and it's impossible to get a good quality.

In the documentation it's stated that Gain and AGC can only be used for Linear modes.

Is this a technical limitation or can these be implemented for non-Linear modes as well?

Testing with SDR++ I initially get the same bad quality, but there I can activate Tuner AGC or raise the Gain to get very good quality.

// Peter

jonsnow1357, 2025-08-04:
you should be able to enable RTLSDR AGC in the config file ... check docs/SDR/rtlsdr.md

darksidelemm, 2025-08-08:
I'm not sure that librtlsdr exposes it correctly, but many of the RTLSDRs (e.g. the RTLSDR v3) do have the same tuner as is in the airspy - the R820T2 (or equivalent), so finer control over the tuner gain settings *should* be possible. (of course the 8-bit ADC is still a bit limitation)

I vaguely recall someone did make a fork of librtlsdr to enable this, but it never ended up in mainline, and so we're stuck with the single gain setting.

closed 2025-10-05 by ka9q