ka9q/ka9q-radio #170: Hydrasdr support request

Joanvoo, 2025-07-25:
Would it be possible to add HydraSDR support?
It looks a lot like the AirSpy R2.
I'd be happy with it.
https://github.com/hydrasdr

kng, 2025-07-29:
it should be possible to base it on the airspy driver iirc.

argilo, 2025-08-04:
Here's a prototype of HydraSDR support:

https://github.com/argilo/ka9q-radio/tree/hydrasdr

Since libhydrasdr is almost identical to libairspy, hydrasdr.c is basically the same as airspy.c, but with `airspy_` replaced with `hydrasdr_`. The deprecated `airspy_init` and `airspy_exit` functions do not have equivalents in libhydrasdr, so I removed those function calls.

argilo, 2025-08-07:
Despite the similarity, HydraSDR is not compatible with libairspy. Even though the factory firmware (version 1.0.0-rc2) uses the same VID & PID as Airspy, `airspy_list_devices` fails to enumerate the HydraSDR because its serial number format is different. (As an experiment, I was able to work around this by changing `STR_PREFIX_SERIAL_AIRSPY_SIZE` from 10 to 12 in libairspy.)

Some software packages (e.g. Gqrx) do not use `airspy_list_devices`, and work out of the box with HydraSDR firmware 1.0.0-rc2. However, as of firmware version 1.0.2, HydraSDR uses its own VID & PID, further breaking compatibility.

Some other differences I have found between libairspy and libhydrasdr:

* The deprecated `airspy_init` and `airspy_exit` functions do not have equivalents in libhydrasdr.
* The `airspy_config_write` and `airspy_config_read` functions do not have equivalents in libhydrasdr.
* The `airspy_r820t_write` and `airspy_r820t_read` functions are renamed to `hydrasdr_r82x_write` and `hydrasdr_r82x_read`.
* libhydrasdr has a `hydrasdr_set_rf_port` function to select between its three RF inputs (one external, two on the circuit board).
* libhydrasdr adds a `hydrasdr_reset` function.

Given the small list of differences, it wouldn't be difficult to make a single library that works with both devices. But I expect the two vendors don't intend to collaborate on software.

Since there are two separate libraries, I thought it would be best to maintain hydrasdr.c separately from airspy.c.

ka9q, 2025-08-08:
Wow! Came right up after I installed argilo's patches. Thanks for your help!

Joanvoo, 2025-08-08:
Thanks argilo, now auto-rx is also working!

ka9q, 2025-08-08:
Rather than guessing, can you tell me? People frequently ask my opinion of what SDR they should buy, and I want to give them an informed opinion. So far this one looks good, especially if their claim to be made in the USA is really true as I can't believe that all the *parts* inside won't be affected by Trump's tariff bullshit. Airspy US pulled out of Dayton this year because the Chinese wouldn't ship them their own units.

bvernoux, 2025-08-10:
> differences

You have forgotten also libhydrasdr hydrasdr_set_freq  requires const uint64_t freq_hz instead of const uint32_t freq_hz.

bvernoux, 2025-08-10:
> Despite the similarity, HydraSDR is not compatible with libairspy. Even though the factory firmware (version 1.0.0-rc2) uses the same VID & PID as Airspy, `airspy_list_devices` fails to enumerate [...]

libhydrasdr hydrasdr_set_freq  requires const uint64_t freq_hz instead of const uint32_t freq_hz.

ka9q, 2025-08-10:
I'm not sure I want to ask about the back story, I'm getting the impression it is rather messy.

I'm just happy there's another SDR front end that ka9q-radio can use, and I want to support both devices even-handedly. It'll be disappointing if libairspy goes non-open-source, but as long as the source is available and can be compiled and installed to work with ka9q-radio, it's tolerable. Of course the old versions of libairspy remain open source for all time, so any important changes in functionality could always be backported.

I don't really see very many possible improvements as libairspy has been stable for some time, and I bypass much of its functionality anyway. E.g, I grab the raw 20 MHz real A/D samples, bypassing the rather inefficient halfband real->complex conversion filter. I wish I could get even more raw access to SDR devices, especially the tuner synthesizer parameters so I could program them directly myself. And MOST IMPORTANTLY, know the EXACT frequency being used.

argilo, 2025-08-17:
HydraSDR support depends on libhydrasdr, which is part of the [rfone_host repository](https://github.com/hydrasdr/rfone_host). If the library is not present, then HydraSDR support will not be built.

jonsnow1357, 2025-08-17:
damn ... more documentation work 😲

PR #173 (2025-08-08): Add support for HydraSDR RFOne
Closes #170.

Here I have added support for HydraSDR RFOne, which is optionally compiled if libhydrasdr is installed or `make HYDRASDR=1` is specified.

Because libhydrasdr is nearly identical to libairspy, the new hydrasdr.c is mostly a copy of airspy.c, with the `airspy_` prefix replaced with `hydrasdr_`.

I did a quick test receiving FM radio, and ka9q-radio ran fine with the HydraSDR. I tested firmware versions 1.0.0-rc2 and 1.0.2.
closed 2025-10-05 by ka9q